iNavFlight/inav #6818: barometer filter unusable

JulioCesarMatias, 2021-04-11:
in the file `navigation_pos_estimator.c` the baro reading filtered in 1hz is never used, the pt1FilterApply3 function returns the filtered value, but it is never used.

https://github.com/iNavFlight/inav/blob/7d24ebd5f75470526f0fba45b890789454b6b36a/src/main/navigation/navigation_pos_estimator.c#L328

DzikuVx, 2021-04-12:
`pt1FilterApply3` returns but also modifies the passed filter state as it's passed as a pointer

closed 2021-04-12 by DzikuVx

JulioCesarMatias, 2021-04-12:
to obtain the filtered value in 1hz of the baro, it would be necessary to get the direct value in the function `pt1FilterApply3`, for example,` posEstimator.baro.alt = pt1FilterApply3 (& posEstimator.baro.avgFilter, posEstimator.baro.alt, US2S (baroDtUs) );` , or use the `posEstimator.baroa.vgFilter.state` pointer.See and test the sketch I passed on for testing >> https://pastebin.com/KNHMZF61

JulioCesarMatias, 2021-04-12:
[image]

using INAV method

JulioCesarMatias, 2021-04-12:
[image]

the correct method

JulioCesarMatias, 2021-04-12:
`posEstimator.baro.avgFilter` is used only in the pt1 filter functions, but it is never used in other functions in order to obtain the `.state`.

I can't see that it works

digitalentity, 2021-04-14:
That's a remainder of AGL estimation. Indeed, curretly the LPF'ed value of the baro altitude is never used and that's WAI.

JulioCesarMatias, 2021-04-14:
will it be removed in the future? Why does alt-hold not work, it’s too slow