iNavFlight/inav #6832: INAV 2.6.1 configurator bugs

Cutbangrian, 2021-04-14:
Hi, I just installed inav 2.6.1 on my F722 wing.

But when it comes to accel calibration, the value keeps on 4096.

I've re flash firmware, re install the configurator, but it keep the same.
I also trying another usb cable. But it just wont work.
Never had this experience before.
[image]

Cutbangrian, 2021-04-14:
ATTENTION...

The problem is, when step 1 is completed then it will turned into blue.
But when I press the button to the step 2, 

The step one go back to grey colour.

Never had this experience before. Please help

This is the pict
[image]

Jetrell, 2021-04-14:
Did you try setting the controller to `Defaults` in the CLI ?  
Then remove the `ACCZERO` and `ACCGAIN` settings from your Diff files (# master) before you reload them into the CLI.

After this you should be able to complete the calibration.

Cutbangrian, 2021-04-15:
> Did you try setting the controller to `Defaults` in the CLI ?
> Then remove the `ACCZERO` and `ACCGAIN` settings from your Diff files (# master) before you reload them into the CLI.
> 
> After this you should be able to complete the calibration.

This stil not clear to me, I'm sorry.
I didnt touch anything in CLi

I just bought this fc and installed the latest firmware 2.6.1 with inav configurator 2.6.

The first thing I did is accel calibration.
But just like I described before, the value wont changes.

If value on gui changes, I can entry the value manually into cli.

But it's not. The value remain 4096 even in gui and cli.
Means not calibrated.

So I need to delete acc zero and acc gain in diff file?

Jetrell, 2021-04-15:
If you haven't started setup. Click the `Reset Settings` button on the configurator setup page.
Then try to run the calibration again.

Cutbangrian, 2021-04-15:
thanks. solved. really appreciate it

closed 2021-04-15 by Cutbangrian