iNavFlight/inav #6833: OSD stats page showing crossfire info when crossfire not used

breadoven, 2021-04-14:
I'm getting Crossfire stats for min RSSI on the new disarm OSD Stats page 1 (V 3.0.0 build). It shows, LQ and RSSI value in dbm, always 0. However I use IBUS which previously showed the min RSSI value as a percentage (IBUS Error value).

OSD.c has a `#if defined(USE_SERIALRX_CRSF)` in:
https://github.com/iNavFlight/inav/blob/34c91848fc984dbc7486a27f76eaf46c60ba86b9/src/main/io/osd.c#L3126

which surely should preclude the crossfire stats being displayed if Crossfire isn't used. So why is it displaying Crossfire stats ?

OptimusTi, 2021-04-14:
Oops! I found a solution and tested between crossfire and SBUS and it's working. On my way to the pub. Looking at the OSD CMS too since it's showing the same issue. I'll post a PR later today.

OptimusTi, 2021-04-15:
Hello @breadoven,

I submitted a fix on #6838 Please check it out. I can only test crossfire and SBUS.

breadoven, 2021-04-16:
OK thanks for that, seems to work fine. I guess it isn't a Flysky IBUS thing as such because "RSSI" is just sent to the FC by an Aux channel in my case.

Don't suppose you fancy adding the date after "STATS" on page 1 ... for those of us who forget to turn on the DVR before arming, e.g. `"-STATS-(date) 1/2 ->"` ?

Or date could be added as one of the extra stats fields together with the time would be even better.

One issue that did occur to me is if you forget to move to page 2 before disconnecting the battery you lose any recording of stats on page 2 whereas before it was all on one screen so always got recorded regardless. Not sure how to get around this other than perhaps to have the screens auto swap back and fore on a timer until the Roll stick is used to swap.

OptimusTi, 2021-04-17:
Ok thanks. The date is in the arming screen along with other good info for recordings.

closed 2021-04-17 by breadoven