iNavFlight/inav #6836: Telemetry issues with Inav & FC-F411 WSE

Viper1200, 2021-04-15:
as per previous reports back in April 2020 by various users I can confirm that the issue still exists.

I am using a Jumper T16 (latest software & firmware) a Radiomaster R168 (latest V2 firmware) and a Mateksys FC F411 WSE.

Whilst the FC appears to control flawlessly the telemetry from the F411 continually drops out, the telemetry from the Rx is fine.

I have tried Motive RC R8SB, Radiomaster R168 & Frsky X8R and all have the same problem dropping some or all telemetry after a short while (several seconds only usually) and then sometimes reconnecting to a few of the sensors before dropping again.

Any assistance would be appreciated

Viper1200, 2021-04-21:
Further information re the Telemetry issues, I am using iNav Configurator V2.6.1 and there appears to be a discrepancy between the GUI and the CLI with regard ti TELEMETRY.
With TELEMETRY turned ON in the GUI the CLI DUMP shows telemetry_switch = OFF, while DIFF does NOT show it at all.
If you set telemetry_switch = ON within the CLI it then shows up as ON in both the DUMP & DIFF file HOWEVER the TELEMETRY function then no longer works at all.  
With the CLI showing telemetry_switch = OFF and the GUI showing Telemetry ON the TELEMETRY function does work but it is a bit sporadic and does NOT appear to send data continuously back to the radio. A lot of the time, whilst the Rx telemetry is scanning fine, the FC telemetry will completely stop scanning & then when a Telemetry Fail alarm comes up it will start scanning again, only to drop out sometime later.

Hope someone with far more knowledge than me can look at the code around the telemetry issue and resolve it.  The Flight Controller appears to work flawlessly but I really only purchased it to get the telemetry back from my plane.

Viper1200, 2021-06-04:
Thought I may have got a response on this, I guess it's just me having the hardware issue.

Mateyhv, 2021-06-10:
No easy answer but my guess is there is something flawed with the way the STM32F411 is used on the WSE board regarding softserial. A friend of mine has had lots os issues setting the SS for telemetry. Sometimes it worked, others did not... totaly random, kind of what happens in your case. My suggestion is to get rid of that board or ask for replacement. Unfortunately the F411 wing is no long into production the twin SS work flawlessly.

closed 2022-01-28 by DzikuVx