iNavFlight/inav #6843: Cruise mode for multirotors

avsaase, 2021-04-16:
For longer range cruise flights it would be nice to have a real cruise mode for multirotors like we have on fixed wing. I propose the following controls:

- When enabled, this mode works just like position hold so the quad hovers in place.
- Push the pitch stick forward to start forward flight at a fixed speed constrained by the cruise speed and the max bank angle. The stick can then be centered again. The quad maintains altitude and heading.
- Pull pitch stick back to go back to a hover.
- During forward flight, roll input makes a coordinated turn.
- Throttle controls altitude just like in position hold.

WDYT?

rts18, 2021-04-16:
A multi-rotor can be flown in `Airmode + Althold + Headinghold`. Its the closest we can get to Cruise Hold at present.

_I think it would be useful to have it all combine into a flight mode._

Although I have some concerns. Have you read my tests and view in #6777 ? This has many similarity's to some of your function suggestions. And the bug is almost certainly going to show up in this feature, unless it is fixed.

JulioCesarMatias, 2021-04-17:
@rts18 I agree

closed 2024-03-20 by b14ckyy