iNavFlight/inav #6851: WP Mode Set Head doesn't automatically reset when starting new mission

breadoven, 2021-04-19:
Set up a mission with a Set Head WP that remained set until the end of the mission. Flew the mission twice without disarming but with WP cancelled and restarted between flights. Noticed that Set Head was still active when starting the second flight. Looking at the code there doesn't appear to be anything that cancels `wpHeadingControl.mode` when a new mission is started which would explain why it remained active.

Probably just needs `wpHeadingControl.mode = NAV_WP_HEAD_MODE_NONE;` setting in:

https://github.com/iNavFlight/inav/blob/c63aa64ff40f9f51d2cfd791be5490b66a53ebb3/src/main/navigation/navigation.c#L1463

stronnag, 2021-04-19:
Indeed. Please PR.

breadoven, 2021-04-19:
Tested and seems to do the job.

closed 2021-04-19 by stronnag