iNavFlight/inav #6859: Should Failsafe trigger after disarm when Rx loss caused unintended rearm ?

breadoven, 2021-04-20:
I recently landed, disarmed the plane and then mistakenly switched off the transmitter before unplugging the plane battery. It went into failsafe and tried to run the motor probably because it had landed more than the RTH minimum distance away from home. The prop was stuck on the ground and stalled so I raised the nose to let the motor run then had to set the Tx switches so it would transmit properly before finally stopping the motor.

I realised after looking at the logs it had rearmed when the Tx was switched off because the arming Aux channel changed to a value that triggered arming. This was partly my mistake because the arming range was way too wide (now reduced). However, it doesn't seem correct that this should happen in the first place, i.e. failsafe should never trigger a RTH if the craft has been intentionally disarmed. INAV knows the disarm was intentional if the RX signal was good when it happened, it also knows when the Rx signal became suspect enough to trigger a failsafe so should block rearming if it occurs at the time the failsafe was detected since it's likely caused by unpredictable behaviour of the Aux channels when the signal is lost. This doesn't seem to happen at the moment ... or is there some setting available that would prevent this ?

INAV Version 3.0.0

rts18, 2021-04-22:
@breadoven 
 Some form of ` flying/not flying ` detection along with your suggestion would be best. A simple check if the GPS speed is under a set threshold should suffice in this case. 

Implementing it as a result of your observation, without ` flying/not flying` detection may cause other problems for users.

* If the receiver failsafe arming channel was set to the disarm position. 
* If failsafe mode test is used.

breadoven, 2021-04-22:
I had a look at this further trying to understand the timings. There's something odd with the timing for setting a blocker "ARMING_DISABLED_RC_LINK" to prevent rearm. You can get things to work as required with relatively large delays (100's milliseconds) on arming (and disarming) but it's not clear why this is necessary because from measured timings an invalid signal is reported to failsafe in a few microseconds. When a flag is set based on this it doesn't work as expected especially for setting the arm blocker. Need to work out where the extra time is coming from. Nothing to do with Failsafe delay settings because I'm ignoring that, being bypassed. "ARMING_DISABLED_RC_LINK" doesn't get set with the current logic because it relies on the delayed failsafe flag for Rx loss which is way too slow (seconds) to block an Aux channel that's changed value during signal loss.

This is using a Flysky radio with failsafe triggered by setting the RX throttle failsafe to a pulse < 1000us, so INAV is detecting invalid pulse as the failsafe trigger.

breadoven, 2021-04-22:
I think I've come to the conclusion the time delay is coming from the time it takes the Rx to switch to failsafe after the Tx signal is lost. This is based on debugging that seemed to show INAV was receiving an Arm signal around 200ms before it detected a failsafe signal from the RX. This is after the Tx had been switched off when disarmed. So in the case of it rearming after disarm when the Tx is switched off the Rx is already outputting a "false" Arm value on the Arm Aux channel for 200ms before the Rx goes into failsafe. Hence the need for maybe a 300ms delay to check if an Arm change of state is caused by a failsafe channel glitch. So the solution to avoiding an unintended rearm is to do the same as is done for an unintended disarm during failsafe, i.e. add a time delay, to arming in this case and also a blocking flag that triggers immediately a failsafe is detected. This works in my case if you use > 300ms.

Testing on a multirotor produces the same result, as you would expect, definitely not something you'd want to be close to, especially when you've just turned the Tx off, because it immediately goes into RTH mode. Even if you're at the home location it tries to land.

MrD-RC, 2021-04-22:
I think the problem here is that with some systems, like I believe the FlySky; there is no failsafe bit, so the failsafe outputs the arm and RTH channel values. If this isn’t the case with FlySky, it is with some other systems, so needs to be considered.

I agree for this to work, there needs to be some kind of flying/not flying estimation. https://github.com/iNavFlight/inav/issues/2331

rts18, 2021-04-22:
> 
> 
> So the solution to avoiding an unintended rearm is to do the same as is done for an unintended disarm during failsafe, i.e. add a time delay, to arming in this case and also a blocking flag that triggers immediately a failsafe is detected. This works in my case if you use > 300ms.

That would be the case. I remember reading a chat here between Digitalentity and Tony Cake over the implementation of the Ghost protocol. And the way iNav had to interpret the Ghost's RX failsafe. 
This took 300mS because Ghost didn't output failsafe data.

breadoven, 2021-04-23:
There does appear to be an issue with the current logic used to prevent an unintended disarm during failsafe which seems to stop it working with Flysky. This is probably the reason I had a wide range set on the Arming channel when Armed. It ensures it remains Armed at the channel value during failsafe but of course causes problems if you're disarmed and have failsafe because it then Arms, as I discovered. The disarm block currently used only seems to look at one check for valid Rx data, it doesn't use the other check which I suspect is needed for Flysky and other similar radios that don't output a specific failsafe flag. Putting a time delay on Arming is probably belts and braces but if it doesn't cause problems it's worth doing, you barely notice 300ms which is what is needed in my case.

I'll just do a PR on this to show what I did. If it's incompatible with other systems or problematic it'll be picked up.

closed 2022-02-10 by DzikuVx