iNavFlight/inav #6873: Autolauch idle throttle delay activation

jaroszmm, 2021-04-23:
## Current Behavior
At the moment if idle in autolaunch is set above 0 motor spins the moment throttle stick is rised.

## Desired Behavior
An user selectable countdown time between throttle stick moved from 0 and idle throttle applied

## Suggested Solution
An option in CLI or GUI to enter desired delay time (10s default) from activating autolauch to spinning the motor. 

## Who does this impact? Who is this for?
Everyone who autolaunches a plane that needs idle running and a firm hold on the plane with two hands.

## Additional context
The idea is simple - instead of picking up the model and using for example your foot to raise the throttle on the transmitter sitting on the ground because I hold the model with both my hands we could have a countdown option in which after raising the trhottle INAV would countdown for an user programmable amount of time before actually spinning the motor in idle. This would allow us to safely activate the autolaunch, pick up the model from the ground, wait for idle and throw.

breadoven, 2021-04-23:
Why do you use both hands to launch ? Is it a big heavy model or do you use the "throw it backwards over your head whilst holding both wings" method to launch a flying wing ?

MrD-RC, 2021-04-25:
This sounds like a great idea, especially for models like the MFD Crosswind (& Mini), XUAV Clouds and Goose. Even the Mini Talon could probably benefit, especially if it's a heavy setup. This video shows exactly why this feature would be warmly welcomed https://www.facebook.com/groups/INAVFixedWing/permalink/1601018246766634/

The only changes I would suggest are: 
- Have the default as 0. So the default behaviour is the same as now, as this would likely be the way most people will use it.
- Something that would be good is a second countdown on the beeper, or the ESC, to let you know how long is left before the idle throttle will kick in. Maybe a double beep at the 1 second mark.

jaroszmm, 2021-04-25:
> Why do you use both hands to launch ? Is it a big heavy model or do you use the "throw it backwards over your head whilst holding both wings" method to launch a flying wing ?

A big model, two handed launch is safer as I can hold it more firmly. Right now I use 0 idle and throw hard hoping for the best, would be nicer to have some help from the idle.

breadoven, 2021-05-09:
Another method rather than using a timer could be to have double launch detection. Arm and move the stick, nothing happens, then do the first "dummy" launch which gets the motor running at idle followed by the second actual launch. Gives more control over the situation and avoids panic if you screw up and run out of time.

jaroszmm, 2021-05-10:
Seems I wasn't the first with this idea:

https://github.com/iNavFlight/inav/issues/5661

avsaase, 2021-05-10:
Then lets continue the discussion there.

closed 2021-05-10 by avsaase