iNavFlight/inav #6894: GPS Time format

Impulze75, 2021-04-27:
## Current Behavior
  GPS time only shows Hour and minutes.

## Desired Behavior
 I want to have Hour/Minutes/Seconds diplayed on OSD for timestamp purpose.

## Suggested Solution

 Just change format

## Who does this impact? Who is this for?
 All Inav user that want a more accurate timestamp overlay in video.  

## Additional context

rts18, 2021-04-27:
@Impulze75 Adding seconds would be best to have configurable.  
Not everyone requires seconds. Having it added will take up more screen space if a user prefers to have time located at the sides.

Impulze75, 2021-04-27:
Yes, thats configurable would be perfect 🙂 Betaflight have seconds when using GPS time, but i prefer using inav 😊

/Mattias

Skaffa Outlook för Android<https://aka.ms/ghei36>

Impulze75, 2021-05-18:
So how do i get a inav version that supports seconds?  

Regards, Mattias

Jetrell, 2021-05-19:
> 
> 
> So how do i get a inav version that supports seconds?

@Impulze75 You can add the pull request yourself and see if a developer in interested in merging it.
Or in the meantime if you are able. Port that part of the code from BetaFlight and make your own iNav build with it added.

> 
> 
> I want there will have date OSD also.

@ghostkcleung `Years/Months/Day` and `Hour/Minutes/Seconds` can be found at the bottom of the OSD arming page.

Impulze75, 2021-05-19:
> > So how do i get a inav version that supports seconds?
> 
> @Impulze75 You can add the pull request yourself and see if a developer in interested in merging it.
> Or in the meantime if you are able. Port that part of the code from BetaFlight and make your own iNav build with it added.
> 

I guess i could do it myself, but is there any "how to" links for doing own builds?

/Mattias

breadoven, 2021-05-19:
> I guess i could do it myself, but is there any "how to" links for doing own builds?
>
> /Mattias

Build your own firware:
https://github.com/iNavFlight/inav/tree/master/docs/development

I could have a look at adding seconds. As said before though it increases the length of the time field which most users probably wouldn't want. However, if the seconds are optional this isn't such an issue.

Impulze75, 2021-05-19:
> Build your own firware:
> https://github.com/iNavFlight/inav/tree/master/docs/development
>
> I could have a look at adding seconds. As said before though it increases the length of the time field which [...]

That sound awesome! I have a specific need to timestamp using inav in a remote controlled fullscale car.  Im only using the inav for OSD purpose with the GPS.

/Mattias

closed 2021-05-21 by DzikuVx