iNavFlight/inav #6903: Max Allowed Altitude

JacLavi, 2021-04-29:
In many (most of) countries rules and regulations exists about maxi altitude depending on the zone you fly. Here (FR_Fr) I believe absolute maximum is 150m. In my yard 25m. Most of the places even in the country the limit is 50m. 

## Current Behavior

Nowadays there are no parameters limiting the altitude the UAV can reach.

## Desired Behavior

Add Inav a mean to forbid the UAV to climb over a limited altitude

## Suggested Solution

Add a parameters the user can set via the Configurator.
 

 
## Who does this impact? Who is this for?

## Additional context

stronnag, 2021-04-29:
Duplicate of #62, #4798, #5998 (at least)

JacLavi, 2021-04-29:
Geonfence #4798 sounds good to me. Consider closing this one.

JacLavi, 2021-04-29:
Redundants tickets ask for very sophisticated - maybe complicated - features. Will you just consider a (not so) simple max_alt ?

closed 2021-05-05 by DzikuVx