iNavFlight/inav #6911: WP mission resume or restart

pitbuild, 2021-05-02:
## Current Behavior

When switching out of a WP mission in another mode and back to the mission, it restarts the mission and puts WP 1 as next WP. 
## Desired Behavior

I would like to choose with a cli command if I want the mission behaviour to restart (currently) or resume and continue with the latest WP it was heading to. 
## Suggested Solution

Set a variable for the latest WP and if chosen in cli to resume, it will take over the WP from the variable 

## Who does this impact? Who is this for?

For all users, plane and copter 
## Additional context

moggiex, 2021-05-05:
I've had this happen, flicked out of WP mode and then the model has then gone back to restart again, which could be quite a number of WP's away.

A timer value would work best here. So for example this would be default, disabled

`nav_wp_switch_grace = 0`

Then say a value of 5

`nav_wp_switch_grace = 5`

To allow the pilot a 5 sec window to flick back into WP mode and for the model to carry on to the next WP.

When this has happened to me, I've flicked back to WP mode almost immediately, but obviously too late. Most likely the same for the OP too.

Matt

pitbuild, 2021-05-05:
My initial thought was, to create a mission that brings me out to a point of interest, where I can switch out of the mission, fly around in another flight mode and after that switch back to the mission (resume) which brings me back home on a wp guided way, not in a straight line like RTH.
I got this idea on a flight, where I wanted to explore a river dam further out, but a straight direct flight line would bring me too close to an airport.

breadoven, 2021-05-06:
A cursory look at this says it's not so difficult to get it to start at whatever WP number you want. The problem is how you tell it to resume at the last WP or start afresh from the beginning of the mission considering it's controlled using just the one mode switch.

breadoven, 2021-05-07:
This could use a switch toggle method to indicate resume, OFF-ON-OFF-ON < 0.5s = resume, Would work but messy.

The other method that could work is to use a stick command whilst in WP mode such as full pitch up (No RC override in WP mode so this won't cause a problem). The stick command would be held at the time the WP mode switch was selected OFF but WP mode would only end when the pitch stick was released back to centre. This would flag resume when WP mode was selected again.

avsaase, 2021-05-07:
The simplest but least flexible way is to just create a cli setting `nav_wp_resume_mission` that toggles between the two behaviors.

breadoven, 2021-05-07:
`nav_wp_resume_mission `could be more than a toggle. If set >= 0 it defines the starting WP. So 0 will start from the beginning (default), 3 resumes from WP 3. If you set it to -1 it would resume from the last WP. Could also include a timer to resume at the last WP regardless of this setting if you accidentally switch out of WP mode and back in quickly enough.

breadoven, 2021-05-07:
Well the switch toggle method works although needs close to 1s to be reliable. So resuming works OK although Jumps need more testing to be sure they work as expected (should do if they aren't reset on resume, just start where they left off).

Considering you probably only want to run a mission once per flight perhaps it's more sensible to make resume the default and restart an option changeable via CMS as above as suits before a flight. The switch toggle method could be used as a backup method to override the setting in flight.

breadoven, 2021-05-08:
Final version using just a setting with:
START - starts from first WP
RESUME - resumes from last active WP
SWITCH - toggles between START and RESUME each time WP mode is selected

Switch allows you some means of changing the behaviour in flight although once toggled to START it's no longer possible to RESUME to anything other than the first WP.

pitbuild, 2021-05-09:
Guys, seriously, that is a perfect solution! I like it a lot!

breadoven, 2021-05-12:
There is a more flexible option to resuming missions which is to use Poshold to change how WP restarts. The restart option could remain but just RESUME ON or OFF. On a mission being interrupted the default would be to resume. If you wanted to restart from the first WP then Poshold would be selected first then WP. WP has priority over Poshold so this method fits. Using Poshold would really be a fall back method to restart at the beginning without having to land.

Would work but perhaps too convoluted having to remember the mode sequence ?

pitbuild, 2021-05-12:
Yeah, I think that might get confusing at the end. Tbh for me only 2 versions which I would have to select in cli before I go flying would be already good enough! The SWITCH option I think is already a good add on.
Simple is most of the time the better

closed 2021-11-07 by breadoven