iNavFlight/inav #6926: Auto mag alignment

nwpulver, 2021-05-06:
## Current Behavior
Currently you must figure out what direction your mag is oriented on your GPS and which axis is which relative to a quad or wing. It is often confusing to figure out and the documentation is not great in my opinion. 

## Desired Behavior
Press a button in Inav (or stick command) to auto align mag on all axis. Something similar to accelerometer calibration. 

## Suggested Solution
Maybe something like point quad or wing N then E then S then W. Repeat inverted and tilted on pitch/roll 45 degrees or something. May be a bit long but IMO would still be easier than trying to figure out correct alignment manually, especially when the preset alignments do not work, which has been every time for me. 

## Who does this impact? Who is this for?
Would be convenient for all users except maybe experts on mag alignment. 

## Additional context
None

julianschweizer, 2021-05-16:
In my opinion mag calibration is the main reason why people go one step away from (one of the best FC firmware) INAV. Should be added prio 1 definitely.

ivirtex, 2021-05-16:
Yes, mag calibration definitely has to be improved. Especially calibrating via stick commands, the only way to tell that calibration has started is to look at the MCU status LED, which not all FC's have.

avsaase, 2021-05-16:
Two things are being confused here. @nwpulver is asking about automatically determining the correct alignment of the magnetometer relative to the flight controller. @julianschweizer and @ivirtex, you are talking about the calibration of the magnetometer. These are not the same thing.

nwpulver, 2021-05-17:
> Two things are being confused here. @nwpulver is asking about automatically determining the correct alignment of the magnetometer relative to the flight controller. @julianschweizer and @ivirtex, you are talking about the calibration of the magnetometer. These are not the same thing.

I think the calibration is not a problem. It’s easy enough to spin the drone around a bit and make sure everything’s working. The alignment relative to FC is what I thought was very confusing. I’m glad to see others feel similarly

closed 2022-02-10 by DzikuVx