iNavFlight/inav #6931: TBS Crossfire Telemetry Sensors Problem in INav

hprompt, 2021-05-07:
Hi There,
First off, thanks for all the Great work that you all do.

I have an issue with the TBS Crossfire Full TX & Diversity Nano v2.0 RX with Inav version 2.61. I’m currently using a HolyBro Kakute F7 V1.5 FC , the CRSF nano RX's TX & RX are connected to UART6 per the manual. My Taranis X9d+ discovered only 10 sensors at first, which are:1RSS,2RSS,RQly,RSNR,ANT,RFMD,TPWR,TRSS,TQly,TSNR. On the Nano RX I turned on the “RX Batt Sensor” then deleted all sensors and discovered 4 new ones, which are: RxBt,Curr,Capa,Bat_. So now I get a total of 14 sensors. I also have a Beltian BN-880 GPS connected to UART3. I’m missing the following sensors: FM,GPS,GSPD,Hdg,Alt,Sats,VSpd,Ptch Roll,Yaw.

I’m referencing the following closed issue for my issue.
TBS Crossfire Telemetry Sensors Problem #5877

Thanks in advance

Hayward

hprompt, 2021-05-16:
how do I label this and get it assigned

marcusmcelroy, 2021-05-19:
Are you sure you have your CRSF TX wired to your UART RX correctly? I've had this problem before, and it turned out because there was a poor connection between NanoRX and FC.

OptimusTi, 2021-05-19:
I have the same setup as Hayward and I have no issues. All works out of the box.

marcusmcelroy, 2021-05-19:
Same.. Which is why I was wondering if there is a wiring issue.

hprompt, 2021-05-19:
thanks for the replies
I will check today to verify the connections from the crossfire Diversity Nano v2.0 RX to the holybro kakute v1.5 FC

hprompt, 2021-05-19:
I checked the connections:
crossfire Diversity Nano v2.0 CH1 which is TX connects to UART6 on kakute F7 v1.5  R6
crossfire Diversity Nano v2.0 CH2 which is RX connects to UART6 on kakute F7 v1.5 T6

hprompt, 2021-05-21:
I'm using the TBS Crossfire Diversity Nano RX v2.0
one thing that I noticed a difference between the diversity and the nano is when I run either the crossfire.lua or agent lite
it shows the below.

TBS Crossfire nano RX shows:
Output 1 Crsf tx
Output 2 Crsf rx
Output 3
Output 4 SmartAudio
Output 5 BST SCL
Output 6 BST SDA

Crossfire Diversity Nano RX only shows:
Output 1 Crsf tx
Output 2 Crsf rx
Output 3
Output 4
Output 5
Output 6

OptimusTi, 2021-05-21:
Same as my diversity nano and I don't have this issue. You could set the outputs like the regular nano but it's not going to matter. 

Which version of openTx and crsf are you using?

hprompt, 2021-05-21:
same as you crsf 4.11 & OTX 2.3.11-otx
what radio are you using? taranis x9d+ here

OptimusTi, 2021-05-21:
Same. And now using RM TX16S

hprompt, 2021-05-21:
so you get all 23 sensors on both x9d+ and RM TX16S radios?
I have a betaflight 4.2.9 build using the same x9d+ radio and get all the sensors, I'm going to compare the DUMPs and DIFFs

OptimusTi, 2021-05-21:
All   24   sensors on both radios, yes.

hprompt, 2021-05-21:
HolyBro Kakute F7 V1.5 FC with CRSF Diversity Nano connected UART6 
Beltian BN-880 GPS connected to UART3, using 3.3V and the SDA/SCL on the Kakute F7

dump

# version
# INAV/KAKUTEF7 2.6.1 Feb 25 2021 / 10:28:17 (64249b1a)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# safehome
safehome 0 0 0 0
safehome 1 0 0 0
safehome 2 0 0 0
safehome 3 0 0 0
safehome 4 0 0 0
safehome 5 0 0 0
safehome 6 0 0 0
safehome 7 0 0 0

# logic
logic 0 0 -1 0 0 0 0 0 0
logic 1 0 -1 0 0 0 0 0 0
logic 2 0 -1 0 0 0 0 0 0
logic 3 0 -1 0 0 0 0 0 0
logic 4 0 -1 0 0 0 0 0 0
logic 5 0 -1 0 0 0 0 0 0
logic 6 0 -1 0 0 0 0 0 0
logic 7 0 -1 0 0 0 0 0 0
logic 8 0 -1 0 0 0 0 0 0
logic 9 0 -1 0 0 0 0 0 0
logic 10 0 -1 0 0 0 0 0 0
logic 11 0 -1 0 0 0 0 0 0
logic 12 0 -1 0 0 0 0 0 0
logic 13 0 -1 0 0 0 0 0 0
logic 14 0 -1 0 0 0 0 0 0
[... 1263 lines cut ...]
set yaw_rate = 60
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 2

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

# end the command batch
batch end

#

OptimusTi, 2021-05-21:
Test serialrx_halfduplex = AUTO

Delete all sensors and rediscover.

That's what I have

hprompt, 2021-05-21:
ok will try and let you know

hprompt, 2021-05-21:
receiver tab changed serialrx_halfduplex = OFF to serialrx_halfduplex = AUTO
Delete all sensors and rediscovered - same 14 sensors

on taranis x9d+ on Hardware page serial port = OFF 
tried Telem Mirror and Telemetry In - same 14 sensors
didn't try LUA, Sbus Trainer or LUA

OptimusTi, 2021-05-21:
Hmm. not sure then. Try deleting the model in OpenTX and setup again. You don't have to change anything excep channel setup, sensor discovery.

hprompt, 2021-05-21:
when you say channel setup that's the input page on the taranis?  Right
So I will delete the model and setup completely again

thanks for all your time and help

hprompt, 2021-05-22:
deleted existing and created new model on taranis x9d+, erased kakute F7 v1.5 FC installed BF 4.2.9 with existing hardware.
deleted and rediscovered sensors same 14 sensors. swapped Diversity nano RX with nano RX same 14 sensors. Flashed kakute F7 with inav 3.0 RC no change same 14 sensors.

Broncas97, 2021-05-29:
Hello guys, I'm having the same problem as the OP. I can only get 13 sensors working. I tried flashing an older version of Crossfire but same problem. I'm on OpenTX 2.9.11 using RS TX16s.

wx4cb, 2021-05-29:
it's s etup problem, they're working fine on multiple birds here with the latest crossfire alpha and inav 3.0 rc2

Broncas97, 2021-05-30:
I should mention that I'm not using iNav, but please leave this reply in case anyone like me has this problem too, this is the only thread I could find talking about this.

THE SOLUTION FOR ME: Updating BetaFlight on my quad. I was using 4.2.0 and I installed 4.2.9 and problem solved, I can now detect all sensors without problem.

Hope this helps someone!
It's curious how this bug is present in both iNav and BF.

wx4cb, 2021-05-30:
@Broncas97 that's strange..... i have 12+ birds here all with either inav or betaflight (betaflight 4.2.0 - inav 2.6 or 3.0rc2) and all on crossfire 4.11 without any issues whatsoever, except one that has a bad receiver in it (or antenna).

heck even took my RC car out this morning for a ride around the block on my front porch and that runs crossfire right into the esc (no FC - yet)

OptimusTi, 2021-05-30:
Strange indeed. Specifically since sensors are a thing between the Rx and OpenTX.

wx4cb, 2021-05-30:
> Strange indeed. Specifically since sensors are a thing between the Rx and OpenTX.

yup. you should still get the rx sensors like TSNR, TLQ, RLQ, etc etc

OptimusTi, 2021-05-30:
Also assuming that everyone read the manual and saw this: "Missing sensors
If you got issues finding sensors or only a few appear, make sure you disabled​ Ignore Instances​ in the telemetry settings page of your radio"

hprompt, 2021-05-30:
Hi All,
Does anybody run: 
Holybro Kakute F7 v1.5 with Tekko32 F3 Metal 4in1 ESC Stack?
Crossfire Diversity nano v2.0 RX or nano RX v4.11 and v4.0.8
Beltian BN-880 GPS 
If I run inav 2.6.1, 3.0RC1, Betaflight 4.2.9 I get the same 14 sensors

I would like to compare my dump, diff all against your
Thx
Hayward

OptimusTi, 2021-05-30:
Did you disabled​ Ignore Instances​ in the telemetry settings page of your radio?

hprompt, 2021-05-30:
> Also assuming that everyone read the manual and saw this: "Missing sensors
> If you got issues finding sensors or only a few appear, make sure you disabled​ Ignore Instances​ in the telemetry settings page of your radio"

tried this only got the same 14 sensors

wx4cb, 2021-05-30:
ok so..

1. enabled telemetry in the configuration tab
2. using CRSF and not SBUS
3. HAVE IT WIRED CORRECTLY
4. in the receiver tab you have serial and crsf setup?
5. Have CRSF on the relevent outputs of the crossfire module in the lua script for the receiver ?

hprompt, 2021-05-30:
> * enabled telemetry in the configuration tab
Yes 

> * using CRSF and not SBUS
Yes

> * HAVE IT WIRED CORRECTLY
crossfire Diversity Nano v2.0 CH1 which is TX connects to UART6 on kakute F7 v1.5 R6
crossfire Diversity Nano v2.0 CH2 which is RX connects to UART6 on kakute F7 v1.5 T6

> * in the receiver tab you have serial and crsf setup?
Yes

> * Have CRSF on the relevent outputs of the crossfire module in the lua script for the receiver ?

If I run the tbs agent lite or the crossfire lua 

Crossfire nano RX shows:
Output 1 Crsf tx
Output 2 Crsf rx
Output 3
Output 4 SmartAudio
Output 5 BST SCL
Output 6 BST SDA

Crossfire Diversity Nano v2.0 RX only shows:
Output 1 Crsf tx
Output 2 Crsf rx
Output 3
Output 4
Output 5
Output 6

wx4cb, 2021-05-30:
i havent touched the half duplex stuff on mine, it's whatever the defaults are for inav, half duplex shouldnt be on anyway. the only crsf line in my diff is:

set serialrx_provider = CRSF

hprompt, 2021-05-30:
> also, i would try another serial port too on the fc that is

Its currently on UART6 (because that's what the manual said to use)
open UARTS are: 2,3 and 7

I sent an email to Holybro about this and they didn't offer any help at all

wx4cb, 2021-05-30:
try one of the other uarts.... ive never used that FC at all. i use matek wing fcs in my planes but the firmware should still be the same.

hprompt, 2021-05-30:
ok I will give it a try
thanks for all your help

Hayward

Broncas97, 2021-05-31:
@hprompt try enabling the Telemetry option in the Configuration tab in BetaFlight.

[image]

hprompt, 2021-05-31:
Running Inav

Broncas97, 2021-05-31:
I don't use iNav but if there is a similar setting, try it. Updating betaflight and turning that setting off and on again was what did it for me. Wish you luck man.

hprompt, 2021-05-31:
I will try disable/enable Thanks for the help

CorruptName, 2021-06-01:
None of these suggestions worked for me. What I did to get telemetry working:
1: On Crossfire Agent Lite or LCD, Under "Output Map" set channels 3&4 to Mavlink RX & Mavlink TX.
2: Wire those pins to an unused UART.
3: On ports tab in inav, select UART then set it to Mavlink (telemetry column) - Make sure to use 57600 BAUD.
4: Delete all then Discover New Sensors.

This is the ONLY WAY I have gotten Telemetry working with crossfire on 2 different wings. Never had to do this before.

wx4cb, 2021-06-01:
@CorruptName That's only if you want true mavlink over bluetooth or wifi. this has no effect on the CRSF telemetry

hprompt, 2021-06-01:
Great I will give it a try
Thx

CorruptName, 2021-06-01:
I wasn't suggesting that it should be necessary. This is what I had to do
to get sensors to come through to opentx. Without this solution, I could
not get most of my telemetry data. It *seems* that I can now remove Mavlink
now that it is working and crossfire will carry telemetry again.

hprompt, 2021-06-08:
Hello All

I did as CorruptName suggested:
1: On Crossfire Agent Lite or LCD, Under "Output Map" set channels 3&4 to Mavlink RX & Mavlink TX.
            I couldn't get CH3 to set to Mavlink RX, it would only set to Mavlink TX. CH4 is set to Mavlink RX
2: Wire those pins to an unused UART. 
            I used UART2
3: On ports tab in inav, select UART then set it to Mavlink (telemetry column) - Make sure to use 57600 BAUD.
           I did this
4: Delete all then Discover New Sensors.
           I did this now I get a total of 22 Sensors
1RSS,2RSS,RQly,RSNR,ANT,RFMD,TPWR,TRSS,TQly,TSNR
GPS,GSpd,ALT,SATS,PITCH,ROLL,YAW,Hdg,RxBT,CURR,CAPA,BAT_

I believe I'm missing the FM sensor, how would I get the FM sensor?

THANKS ALL FOR YOUR HELP

Hayward

CorruptName, 2021-06-08:
> Hello All
> 
> I did as CorruptName suggested:
> 1: On Crossfire Agent Lite or LCD, Under "Output Map" set channels 3&4 to Mavlink RX & Mavlink TX.
> I couldn't get CH3 to set to Mavlink RX, it would only set to Mavlink TX. CH4 is set to Mavlink RX
You must be using Diversity Nano (or I have it backwards) on one TX is channel 3 and the other RX is 3. Doesn't matter as long as you connect it correctly.

> I believe I'm missing the FM sensor, how would I get the FM sensor?
So now that you have gotten the telemetry flowing again, you *SHOULD* be able to revert back to just CRSF protocol. You won't get FM using Mavlink, only CRSF.
I suggest you leave channels 2/3 physically connected but disable it in inav & outputs map. Just set outputs 3 & 4 back to Channels 3 & 4. Then Delete/Discover again to make sure everything is working now. If not, you may need to leave Mavlink and do without FM.

CorruptName, 2021-06-08:
> @CorruptName That's only if you want true mavlink over bluetooth or wifi. this has no effect on the CRSF telemetry

It seems there is a bug in inav which is causing CRSF not to work and for some reason Setting up Mavlink gets it working again. Mavlink does not need to stay enabled.

wx4cb, 2021-06-08:
> > @CorruptName That's only if you want true mavlink over bluetooth or wifi. this has no effect on the CRSF telemetry
> 
> It seems there is a bug in inav which is causing CRSF not to work and for some reason Setting up Mavlink gets it working again. Mavlink does not need to stay enabled.

that's strange, I just got done setting up 2 controllers for new planes and didnt have any issues with RC2. can you be more specific about it ?

CorruptName, 2021-06-08:
I don't know much, just that Myself and other users suddenly stop getting
telemetry through crossfire. For some reason enabling Mavlink (temporarily)
fixes the issue. I'll try and reproduce the issue when I get a chance.

hprompt, 2021-06-08:
I’m using a kakute F7 V1.5 FC

CorruptName, 2021-06-08:
I've had this issue on at least 3 different flight controllers lately.
Don't think it's hardware related. Probably need to look at changes to the
radio protocols in 3.0. I do believe the issue is present on both RC1 & RC2
but not certain.

hprompt, 2021-06-09:
I followed CorruptName
I removed MavLink TX and RX from CH3 and CH4 on the Diversity Nano RX output map
removed MavLink 57600 from the Telemetry Column on UART2
Deleted and rediscovered sensors on the Taranis X9D+ and only got the 14 I had before.
I put everything back the way it was and get the 22 sensors again
still missing FM sensor

thanks all

So now that you have gotten the telemetry flowing again, you SHOULD be able to revert back to just CRSF protocol. You won't get FM using Mavlink, only CRSF.
I suggest you leave channels 2/3 physically connected but disable it in inav & outputs map. Just set outputs 3 & 4 back to Channels 3 & 4. Then Delete/Discover again to make sure everything is working now. If not, you may need to leave Mavlink and do without FM.

pilotltd, 2021-06-23:
Tried all of the above with Matek 405 STD and all I can get is 10 sensors! However replacing with a Matek F722, all works fine - exactly same wiring, ports and parameters in crossfire and Inav.

OptimusTi, 2021-06-23:
Which UARTs were you using in the 405?

wx4cb, 2021-06-23:
i literally just wired up an f 405 and had zero issues.

hprompt, 2021-06-24:
I still can’t get the FM sensor to show up, I wanted to run the inav lua telemetry script.other than that I’m ok with my setup.

CorruptName, 2021-06-24:
> I still can’t get the FM sensor to show up, I wanted to run the inav lua telemetry script.other than that I’m ok with my setup.  Sent from Yahoo Mail for iPhone On Wednesday, June 23, 2021, 3:52 PM, wx4cb ***@***.***> wrote: i literally just wired up an f 405 and had zero issues.

Thats because "FM" Is a crossfire telemetry object and not a Mavlink object from what I can tell.

CorruptName, 2021-06-24:
> i literally just wired up an f 405 and had zero issues.

Probably because this is an intermittent problem. I still haven't figured out what actions can reproduce it.

hprompt, 2021-06-24:
That’s right but even if I remove mavlink it still doesn’t show up.

CorruptName, 2021-06-24:
Stupid question, are you deleting & rediscovering sensors in opentx?

hprompt, 2021-06-24:
YesI’ve tried that several time

[20 more replies]

closed 2021-08-23 by hprompt