iNavFlight/inav #6940: Dshot beeper GPS lock issue 3.0

Jetrell, 2021-05-09:
## Current Behavior
Dshot beeper will not work with a GPS fix.
Once a fix is lost. It will commence working again.

## Steps to Reproduce

1. Enable beeper in the CLI
2. Turn On beeper switch and it will work.
3. Once the GPS gets a valid fix. The beeper stops working.
4. Cover the GPS.. Once the Lock is lost, the beeper starts working again.

## Additional context (That may be related to the new Dshot command features)

Once Air mode throttle threshold is met. Disarming the quad will cause the motors to burst into life for a moment and flip the quad over. This effect only occurs if there is a GPS fix. 

- FC Board name and vendor: BLheli_32 Rev - 32.8
- INAV version string: current 3.0 development

avsaase, 2021-05-10:
> Once Air mode throttle threshold is met. Disarming the quad will cause the motors to burst into life for a moment and flip the quad over. This effect only occurs if there is a GPS fix.

So essentially after every flight disarming causes the motors to spin up enough to flip over the quad?

DzikuVx, 2021-05-10:
@kernel-machine any idea?

kernel-machine, 2021-05-10:
I have tested now and i confirm this issue, when the quad gets a GPS fix the ESC beeper doesn't work.
The motors direction spin change with the turtle mode, with or without the GPS fix, so i think that the dshot command part works fine, so maybe the problem is on the dshoot beeper code

kernel-machine, 2021-05-10:
I CATCH IT

DShot beeper works  only with these beeper mode BEEPER_RX_SET or BEEPER_RX_LOST
https://github.com/iNavFlight/inav/blob/449d7929db67aa81377ab249db805d1a5ed0e2ca/src/main/io/beeper.c#L344
And when the BEEPER MODE is triggered and the FEATURE(GPS) is true, this function is called
https://github.com/iNavFlight/inav/blob/449d7929db67aa81377ab249db805d1a5ed0e2ca/src/main/io/beeper.c#L286
And if there is a GPS fix, this function sets the beep mode with BEEPER_MULTI_BEEPS otherwise with BEEPER_RX_SET.
Since dshoot beeper works only with BEEPER_RX_SET and BEEPER_RX_LOST and not with BEEPER_MULTI_BEEPS, when there is a GPS fix it doesn't work

Editing this line 
https://github.com/iNavFlight/inav/blob/449d7929db67aa81377ab249db805d1a5ed0e2ca/src/main/io/beeper.c#L344
to 
`&& (currentBeeperEntry->mode == BEEPER_RX_SET || currentBeeperEntry->mode == BEEPER_RX_LOST || currentBeeperEntry->mode == BEEPER_MULTI_BEEPS)`
it works fine

I don't understand why to use dhsoot beeper only for BEEPER_RX_SET and BEEPER_RX_LOST and not for all beeper mode, is there a specific reason? We have about others 20 modes

PR #6945 (2021-05-11): DShot beeper fix
Fix #6940 so with this PR dshoot bepper works for all beeper modes, as a normal beeper, so on flight mode changing, GPS fix success, ecc...
I has also changed a condition in the activation if

I think that is useless to have a delay after the disarm so i've removed it.

But i've added a delay between the dshoot beeps
closed 2021-05-11 by DzikuVx