iNavFlight/inav #6947: Compass calibration

hidrajulika, 2021-05-10:
Hello, 

Sorry if this is not the right place to ask but I'm struggling with compass, since my module is in angle with the FC.  The compass module is 15 degree tiled to the FC (in order to have a better GPS orientation to the sky in most long-range  flying conditions.) This is a simple 6inch quad.

So after playing around with orientation (flip cw90 works best) and calibrating, the following result I can achieve:

When I point the quad to 0deg (North) it shows 0deg. When I turn it to 90deg it shows 120-is. When I point it to 180 it shows correct 180. The other half is the same non-linearity. 
If I point the quad to 0deg, and pitch it, the heading goes off. (The pitch affect the heading reading.)
I suppose it is due to the tilted module. 

I find no way to fix it.
I tried the mag alignments (flip/nonflip 0,90,180,270),  the above mentioned works best.
I was playing with align_mag_pitch etc values, without luck (no documentation I found on that). 
I tried align_mag_pitch 1950 and 1650 and align_mag_yaw 90, but it was always doing some nonsense.

The module is a BN-880.
The FC is a Matek 405 clone (Eachine)
I tried INAV 2.6.0 and 2.6.1 the results are the same. 

Any advice or a clear info on align_mag_... settings would be welcomed.

breadoven, 2021-05-11:
I found these useful especially the bit where you have the GPS tilted:

https://github.com/iNavFlight/inav/issues/5405
https://github.com/iNavFlight/inav/issues/6232#issuecomment-727636397

 I assume you have the connector to the rear. That's supposed to be the way the BN880 is mounted.

hidrajulika, 2021-05-11:
Hello again,

And thank you breadoven, your references seem to have lead me to solution. 
At least on the bench... , but this should work.
As my BN880 has connectors to the front, with 15deg backtilt the working settings for me are:

`align_mag_roll = 150
align_mag_pitch = 1800
align_mag_yaw = 900
`

breadoven, 2021-05-11:
The BN880 compass module supposedly has an arrow on it but it's hidden by the GPS that's stuck on top. Shouldn't matter though if you allow for it in the settings.

https://www.unmannedtechshop.co.uk/product/beitian-bn-880-gps-compass-module/