iNavFlight/inav #6954: 3.0 CLI Diff_Dump extraction not working on F4

Jetrell, 2021-05-12:
The problems occurs when a `Diff All` or `Dump` is executed with an F405. While the problem does not effect an F722.
This issue only turned up recently. Its not a Configurator issue.

@CapnBry I think it maybe related to #6933

CapnBry, 2021-05-12:
> @CapnBry I think it maybe related to #6933

I've taken a second look at my change and I don't believe this is the case. serialPassthrough() is only activated if you type "serialpassthrough" in the CLI or the same command is received over MSP, so the code should never be executed otherwise. This is on the 3.0.0 branch.

I have two F4 boards here, both with the change (against 2.6.1) and they both can diff all and dump without any problem and I verified that the data is complete as expected. I think you'll have to look elsewhere.

Jetrell, 2021-05-12:
@CapnBry No worries. Thanks for looking into it.
What ever the cause. Its only turned up in the last couple of days.

avsaase, 2021-05-12:
I don't have any issues with a recent build on an F411.

stronnag, 2021-05-12:
If you got the hex files from me, there is the possibility that it's a compiler issue, as the compiler was upgraded here to GCC 11.0 a few days ago .... investigating.

stronnag, 2021-05-12:
Seems like there is a GCC 11 issue (or an inav issue provoked by 11).

* F405, `-DCMAKE_BUILD_TYPE=RelWithDebInfo` :  dump stops, board unresponsive after c. 2000 bytes output
* F405, `-DCMAKE_BUILD_TYPE=MinSizeRel` : board locks up instantly on entering CLI
* MSP seems unaffected.

This may also affect GCC 10.20. I'll look at that later

Back to 9.3 for http://seyrsnys.myzen.co.uk/inav_dev/ builds.

@Jetrell, if you're using a seyrsnys.myzen.co.uk build, please check with the latest update and close the issue if the compiler reversion has fixed it.

Jetrell, 2021-05-12:
@stronnag Thanks for the input. That was the case. 
Sorry for the false alarm people.

closed 2021-05-12 by Jetrell

stronnag, 2021-05-13:
> @stronnag Thanks for the input. That was the case.
> Sorry for the false alarm people.

Thanks for discovering and reporting. We do upgrade the compiler for official builds, so it's useful to have a early indication of issues.