iNavFlight/inav #6956: [Request] Replace Adjustment feature completely with In-Flight Adjustment Menu

b14ckyy, 2021-05-12:
## Current Behavior
Currently we have adjustments we can set up manually to have a specific set of values we can change in flight. But we are always limited by the number of available adjustments and also OSD space.

## Desired Behavior
A Full Menu that can be accessed in flight and controlled with 2 RC channels as it is already the case with the current Adjustments.

## Suggested Solution
The Adjustment Menu could include all possible adjustable values ordered in pages. The Pages could then look like this (excluding the # signs) and replaces the AHI, Crosshair and scroll sidebars and completely takes this 13x10 character area. 15x12 should also be possible but could cover other elements too. 8 Values per Page. Active value will blink.

`###############
#Page 1/X     #
#-------------#
#Roll P    XXX#
#Roll I    XXX#
#Roll D    XXX#
#Roll FF   XXX#
#Pitch P   XXX#
#Pitch I   XXX#
#Pitch D   XXX#
#Pitch FF  XXX#
###############

###############
#Page 2/X     #
#-------------#
#Yaw  P    XXX#
#Yaw  I    XXX#
#Yaw  D    XXX#
#Yaw  FF   XXX#
#RollRate  XXX#
#PitchRate XXX#
#YawRate   XXX#
#R/P Exp   XXX#
###############`

I am thinking of the followinavigation concepts. 

**Concept 1**
We use 2 RC channels with default (neutral) value in the center and should be self centering 3pos switches (trim switches on the radio for example) or alternatively 3pos switches the user has to center manuall. 

CH1 is used to show and hide the menu as well as scrolling and page change. 
CH1 HIGH hold for 2s > shows menu on the OSD
While menu is shown CH1 HIGH for 1s switches to the next page (page number is shown in first line; no reverse page switch)
CH1 LOW hold for 2s > exits menu
CH1 HIGH/LOW Pulse switches between lines in the current page

CH2 HIGH/LOW changes value

**Concept 2**
We still use 2 RC Channels but one has to be a Potentiometer and the second one a 3pos momentary switch.

POT1 between 1000-1200 > Menu is not shown
POT1 between 1200-2000 > selects items on current page in 8 segments
CH2 HIGH/LOW pulses change value
CH2 HIGH/LOW for over 1s > switches between pages (no continous value change)

## Who does this impact? Who is this for?
All advanced users who wants to do fast and precise tuning without the need of landings in between. Especially wing pilots where landing and launching takes more time and effort compared to multirotor and flight times are much longer. 

## Additional context
This is just a first idea. A big advantage would be, to change the adjustable values in the future without the need of changes in the Configurator. This would make future development more easy.

MrD-RC, 2021-05-12:
Another option for accessing the menu would be having a mode. Then it’s up to the pilot how they access. For example, I could change my OSD pot to have this as the last option, just bu changing a mode slider.

b14ckyy, 2021-05-13:
@MrD-RC thats a great idea. I added this as Concept 3.

FPVZaphod, 2021-05-24:
Would be great.
Currently I programmed by Tango2 in such a way. The right 3-Pos switch cycles through the different options by increasing or decreasing (up / down steps) a OpenTX-gVar. This gvar directly controls an RC channel. Center (1500uS) means "Off", where the buttons for increasing / decreasing dont do anything.  With the 3-Pos switch you can choose:   Cruise-Thr / TPA / Board alignment roll / Board alignent pitch / OFF / Roll P / Roll I / Pitch P / Pitch I.  With the two buttons on the rear side of the Tango you can increase or decrease the values.  OpenTX tells you by voice output, which value is currently to change. 

For your idea, I would like to add the possibility to not show a whole menu, only a single line with the value to change. Can be "OFF", or everything you mentioned. Not as beautiful, but this could be used with DJI FPV, where you only have the Craft Name to show custom text.

b14ckyy, 2021-05-24:
that's similar to what our OpenTX Pro Model Preset does. We use one trim switch to raise a Gv and output that to a channel for 20 values to select. and a second trim switch to pulse a second channel for changes. 

but yes a simple line would maybe enough. AS long as it is selectable what items are active in the list so we don't over fill it. 
But to be honest. Just because Of DJI users not having a full OSD I don't see a reason why we should limit analog users ;)  

If it will be a simple list, then maybe we can at least show 5 rows. DJI users will still see the selected adjustment on the name tag as it is now possible with 3.0

closed 2022-02-10 by DzikuVx

b14ckyy, 2022-02-10:
@DzikuVx why did you close this?