iNavFlight/inav #6958: Matek F411WSE Barometer is not detected (stays in red)

alexvazquez, 2021-05-13:
I flashed a Matek F411WSE with the latest Firmware version 2.6.1 and the barometer is not detected. Stays in red color.

[image]

I also tried to downgrade the firmware to other versions and the same thing happened.  I was able to get it detected just once by disabling the barometer under Configurations and save and then going back and assign the BMP280 and save it again and it worked but just once. I haven't been able to get it to work again so seems a bug somewhere.

Jetrell, 2021-05-19:
> 
> 
> Update: If I plug the battery first and then plug the USB I can get the Baro working. This is the only way I can make it work always but its a bug for sure.

Are you sure that you don't have anything connect to the 3.3v supply pad that is drawing beyond the regulators limit?
The Low Drop Out regulator for that supply and baro is only 200mA max.

Jetrell, 2021-05-19:
>
>    Update: If I plug the battery first and then plug the USB I can get the Baro working. This is the only way I can make it work always but its a bug for sure.

Are you sure that your don't have anything connect to the 3.3v supply pad that is drawing beyond the regulators limit?
The Low Drop Out regulator for that supply and baro is only 200mA max.

closed 2021-06-20 by DzikuVx