iNavFlight/inav #6971: Only 2 motors operate

pbreed, 2021-05-16:
Kakute F7 HDV.
Using Dshot 600 to talk to 4n1 ESC.
Using the outputs tab only motor 1,2 operate. 3,4 don't.
The cli resource command shows motor 3,4 in exactly the same place as 2.6.1.

The hardware used for this test worked under 2.6.1 both before the 3.0 test and after the 3.0 test.
****
## Current Behavior
With motores enabled the configurator slider only works for motor 1,2.
Thje slider for 3,4 have no effect and motors 3,4 do not run.

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

Expected all 4 motors to work.

## Suggested solution(s)

## Additional context

---

Target: KAKUTEF7HDV
Name/Version:
Binary: inav_3.0.0_KAKUTEF7HDV.hex
Date: 2021-5-15 17:36
State: release-candidate

- FC Board name and vendor:
KAKUTEF7HDV

- INAV version string:
Target: KAKUTEF7HDV
Name/Version:
Binary: inav_3.0.0_KAKUTEF7HDV.hex
Date: 2021-5-15 17:36
State: release-candidate

DzikuVx, 2021-05-17:
@pbreed The outputh mapping of KakuteF7HDV have not changed between releases. https://github.com/iNavFlight/inav/commits/master/src/main/target/KAKUTEF7/target.c 
Also, it's the same mapping as on KakuteF7 and I'm using this board on two of my quads no problems with DSHOT.
You also failed to provide the dump so there is no way to find out what's going on

pbreed, 2021-05-17:
I'll try again and gather more info later this week.
(Quad was converted back to 2.6.1 to verify it was not a hardware issue before posting)

pbreed, 2021-05-21:
I re-ran the EXACT same toest today and found no fault.
Not sure what I did wrong to begin with, but its working correctly now.

closed 2021-05-21 by pbreed