iNavFlight/inav #6973: 3.0RC1 Gyro changes effecting sub 4" quads

rts18, 2021-05-17:
The gyro changes to 3.0 have massively increased the presence of _prop wash oscillations_ on smaller performance quads.  

The test quad is a 3". That preformed very well in 2.6.  But now has prop wash oscillations that can not be tuned out.
@DzikuVx This issue occurrs after the commits on April 18.  

This particular quad is perfect in mechanical setup, and has no damage to any of its components. It has FC silicon dampers, and has the correct stack tension. I have also double checked everything before this report.

I can load the same PID settings back to back with both firmware's. As seen before the April 18 commits and after. And this is the result.

Both these tests where done in free fall. Then quickly raising the throttle, to slow the descent.
The grey trace on the RC command is the throttle. 

**Before** d50ae85 5d3a01c
[image]
Their is a small amount of PWO seen in 2.6.1, if I raise the throttle very abruptly . But its extremely small.

**3.0RC1** 
[image]

I am unsure how the removal of `gyro_sync` and the addition of `gyro_anti_aliasing_lpf_hz` is effecting this, by sampling faster and possibly receiving more noise. To be honest. I don't totally understand this command and how it works.

But to add to this whole issue. I did not experience any _prop wash oscillation_ issues with my lower power and larger 6" quads, using 3.0. 
So it maybe related to the higher power to weight ratio and higher resonate frequencies on smaller quads. 
I have run this quad on BetaFlight and it doesn't experience _prop wash oscillation_ either.

DzikuVx, 2021-05-17:
The thing is, that filter strategy have not really changed between
releases. It's the same filters in the same order.

Please provide full dump of both situations as well as full bb logs not
only screenshots.

And gyro_sync is irrelevant over here.

rts18, 2021-05-17:
@DzikuVx I don't have access to the PC with the logs ATM. But I have some more screen shots.

**The PID's are identical, between both these flight**. [Diff all.txt](https://github.com/iNavFlight/inav/files/6492148/Diff.all.txt)
I have know about this for when it first occurred. But didn't report it until now. In case there was more to be merged that would fix the issue.

The logs reveal that there is some `prop wash oscillations` with the builds before April 18. But not noticeable in flight or audible.
I was also pushing the throttle much harder in my tests of the Pre April 18th build. To make it get _prop wash oscillations_.

Reducing roll Proportional gain and changing the Derivative made little difference. In fact increasing the D-term on roll from 19 to 24. Made it oscillate even when hovering.

**Before April 18th commits**
[image]

**After**
[image]

DzikuVx, 2021-05-17:
@rts18 I need both dumps. Some names changed and I need to know what is exact configuration that causes oscillations on your side.

rts18, 2021-05-18:
@DzikuVx These are the logs with their dump files. Both 3.0RC1 and before April 18.
[prop wash oscillation tests.zip](https://github.com/iNavFlight/inav/files/6498268/prop.wash.oscillation.tests.zip)

rts18, 2021-05-18:
@DzikuVx I found the fault, once I did a compare. 
I must of had my eyes closed concerning the slightly different wording of the `gyro_lpf_hz `and `gyro_main_lpf_hz`.

Now that I increased the filtering over the peak to 140hz. And not the 60hz default. Its now fine.
I'll keep it at BIQUAD for a  while. If performance doesn't noticeably lack. I wont go back to PT1.

[image]

Thanks for taking note of my issue still :+1:

closed 2021-05-18 by rts18

rts18, 2021-05-18:
@DzikuVx I think this info tab description is what confused me. Is that description backwards?
Should it not be. The Higher the value , the more delay and the less noise gets through, due to filtering. Is that correct?

[image]

DzikuVx, 2021-05-18:
Lower value = more delay and more filtering. 
higher value = less delay, less filtering

This is a cutoff frequency. We cut frequencies (ok, it's a simplification) above cutoff

rts18, 2021-05-18:
Okay, Understood now, Thank you for the clarification :+1: