iNavFlight/inav #6978: GEPRCF22 support

mooiweertje, 2021-05-17:
## Current Behavior
GEPRCF22 not supported

## Desired Behavior
GEPRCF22 supported

## Suggested Solution
add GEPRCF22 support

## Who does this impact? Who is this for?
GEPRCF22 users

mooiweertje, 2021-05-17:
I made an implementation to be tested.
binary
[jnav_2.6.1.4_GEPRCF722.zip](https://github.com/iNavFlight/inav/files/6493212/jnav_2.6.1.4_GEPRCF722.zip)
source
[GEPRCF722.zip](https://github.com/iNavFlight/inav/files/6493214/GEPRCF722.zip)

Sedgee, 2021-05-17:
Thanks for this, I have the HD version of this board, will give it a go and let you know how it goes.

mooiweertje, 2021-05-17:
I don't even know if it will install and run at all but it matches the BF config as as .

mooiweertje, 2021-05-17:
> Thanks for this, I have the HD version of this board, will give it a go and let you know how it goes.

Well I never made an implementation without having the board itself so this might prove to be a challenge. I hope it installs and runs.

Sedgee, 2021-05-17:
Well I never made an implementation so again thanks. I'll take it cautiously and here's hoping.

mooiweertje, 2021-05-17:
Skip the hope, add patience and perseverance and I have no doubt it will work soon.

mooiweertje, 2021-05-17:
This is the board of the GepRC Crocodile Baby 5?
[image]
[image]

Sedgee, 2021-05-17:
Hope suffices till perseverance and patience pays off.
Is that the board you've compiled for?
I'm out at the mo but I'll update my spec when I get home.

[image]<https://user-images.githubusercontent.com/37510531/118480573-b7731900-b712-11eb-91b2-a8f14eb70c26.png>
[image]<https://user-images.githubusercontent.com/37510531/118480756-f2754c80-b712-11eb-8fa3-efaaac0fac3f.png>

mooiweertje, 2021-05-17:
The build is setup the same as BF GEPRCF722 target.

mooiweertje, 2021-05-20:
> Thanks for this, I have the HD version of this board, will give it a go and let you know how it goes.

Is it working?

Sedgee, 2021-05-20:
Not had chance to test due to weather, will be a few days looking at the forecast. Will let you know as soon as I get the opportunity.

mooiweertje, 2021-05-26:
Does it flash and run on the board?

shcho76, 2021-05-26:
Thank you mooiweertje~~!! 
I'm super newbie for drone and inav~~!! 

I flashed your firmware to my span f722 hd fc  and it's run normally!! Great job bros!

I built up for 3inch cinewhoop with gps(without compass) and baro sensor.

I found a issue for the althold, 

When i setup without use hover throttle option. Alt hold function is almost acceptable but the drone reach the sky with full throttle when i choose althold with disable hover throttle option

 I tried to fix it... change some value, set cli commands...Just follow github discussion.. but still reach the moon ~ ~ 

I'm not sure what i did are correct or no correct ^^

mooiweertje, 2021-05-26:
That sounds like the normal iNav config challenge.

Nice to hear the build is working. If any hardware or pad doesn't seem to work I can help you.

shcho76, 2021-05-26:
I'd like to add some sensor(optical-flow, ridar, gps with compass, ). Already ordered matek gps and optocal-flow 

I have wondering about uart port. According to manual there are 5uart available. But i found only 2 vacant uart pin.

The pin diagram of f722 hd version is perfect same as your pic'.. i couldn't find not only uart3(may be share with scl sda pin)but also uart4 pin. 
And only 4 uart detected on BF & INAV configurator.
![port](https://user-images.githubusercontent.com/84860122/119694061-d77ba880-be87-11eb-8986-91c0c29f1849.png

 i requested pin out definition to geprc  fot  soft serial setup but they never answers..

mooiweertje, 2021-05-26:
There are 4 uarts mapped. 1, 2, 4 and 6. The first one in your picture, VCP, is the USB port.

mooiweertje, 2021-05-26:
There are 4 uarts mapped. 1, 2, 4 and 6. The first one in your picture, VCP, is the USB port. The board itself shows pads for 3 but not 4. That is rather odd. But I'm not sure if the board pictures are correct. I wanted to show pictured of the Baby 5 board but I'm not sure what it looks like. I'm pretty sure the mapping in the software is correct for the board in the baby 5 though.

mooiweertje, 2021-05-26:
Here it says the BF target is 'GEPRCF722 AIO' but in BF that target does not exist.So I used GEPRCF722. https://geprc.com/product/geprc-crocodile5-baby-lr-hd-longrange-fpv/

mooiweertje, 2021-05-26:
There is a third target called GEPR-GEPRCF722_BT which has all 6 uarts mapped.

mooiweertje, 2021-05-26:
The one in the picture seems to have pads for uart 1, 2, 3 and 6 which doesn't match any of the three targets.

mooiweertje, 2021-05-26:
If there are no pads for the uarts, softserial will not help you.

shcho76, 2021-05-26:
I looking for crocobile baby fc pic's.. but i can't 

I checked out youtude about croco' baby 4. 
The  f722aio fc looks like 20mmx20mm. So I think the pin diagram of  f722 aio is similar f722 stable series 

Anyway i was so lucky if you had a target firmware of f722 aio for BF. I never get inav firmware !! 😁🤣

robustini, 2021-05-31:
> 
> 
> I made an implementation to be tested. I don't have the hardware just BF build config. Let me know what the findings are and I will make repairs.
> binary: [jnav_2.6.1.4_GEPRCF722.zip](https://github.com/iNavFlight/inav/files/6493212/jnav_2.6.1.4_GEPRCF722.zip)
> source: [GEPRCF722.zip](https://github.com/iNavFlight/inav/files/6493214/GEPRCF722.zip)

Much appreciated your work, thank you!
I have this board, I think it is identical simply remarked: https://it.aliexpress.com/item/1005002006771976.html?spm=a2g0s.9042311.0.0.114f4c4dJpuaQS 
I can compile and test this build, my only problem is that I have the magnetometer obviously connected to SDA / SCL (RX3 / TX3) and I don't know how to remap it in Target.h.
I also have camera control on Motor5 (B06), of course everything works on Betaflight.
If you help me I can test it, thanks!

mooiweertje, 2021-05-31:
The Betaflight B04 is the real digital B04. That pin on the CPU is connected in the FC print with motor pad 1.

mooiweertje, 2021-05-31:
The Betaflight B04 is the real digital B04. That pin on the CPU is connected in the FC print with motor pad 1.
It's strange indeed that the BF I2C 2 binding is not the same pads as UART3. Maybe there is some electronics involved here?

robustini, 2021-05-31:
> 
> 
> The Betaflight B04 is the real digital B04. That pin on the CPU is connected in the FC print with motor pad 1.

Thanks again for the target, tested everything and it seems almost ok!
I don't just work blackbox (says "onboard SD cardd fatal error) and WS2811 led strip, obviously in BF they work.
What could it be?

mooiweertje, 2021-05-31:
Do you use the GEPRCF722 or GEPRCF722BT betaflight firmware?

mooiweertje, 2021-05-31:
Do you use the GEPRCF722 or GEPRCF722BT betaflight firmware?
I just noticed the GEPRCF722BT had the same pin mapping for I2C-2 as UART3.

robustini, 2021-05-31:
I use GEPRCF722, I2C (mag) for fine with your target.

mooiweertje, 2021-05-31:
If I understand correctly you've got a receiver and a magnetometer on the same two pads RX3/TX3/SDA/SCL?

robustini, 2021-05-31:
No, serial rx is on UART2, see my image.
UART3 is only for I2C here.

robustini, 2021-05-31:
[image]

mooiweertje, 2021-05-31:
And the led strip is working too?

robustini, 2021-05-31:
As I wrote above LED and internal flash blackbox doesn't work with your target file.

mooiweertje, 2021-05-31:
And in betaflight the both work on the GEPRCF722 target?
So you don't use the GEPRCF722BT betaflight firmware?

I'm just making sure so I'm sure I know what to look for. The mappings of the different GEPRCF722xxx betaflight firmwares are rather different.

robustini, 2021-05-31:
In Betaflight this board is flashed with the same hex as many other F7s ([betaflight_4.2.9_STM32F7X2.hex](betaflight_4.2.9_STM32F7X2.hex)), and after the flash from the configurator a different config file is sent which performs the correct remap for the various boards.
For example for my board this config is automatic written after the flash: https://raw.githubusercontent.com/betaflight/unified-targets/master/configs/default/GEPR-GEPRCF722.config 

[image]

mooiweertje, 2021-05-31:
yes thank you. i saw something missing in the led strip part. Haven't found what's missing in the flash yet.. I'll make a new build for you later today.

mooiweertje, 2021-05-31:
yes thank you. i saw something missing in the led strip part. Haven't found what's missing in the flash yet.. I'll make a new build for you later today.

thanks for testing!

robustini, 2021-05-31:
I'm checking your target too.h, I can't find where the problem lies.

mooiweertje, 2021-05-31:
For the led strip a timer needs to be defined in target.c.

mooiweertje, 2021-05-31:
For the led strip a timer needs to be defined in target.c.
    DEF_TIM(TIM1, CH1, PA8, TIM_USE_LED, 0, 0),                            // LED STRIP  D(1, 5, 3)

mooiweertje, 2021-05-31:
As far as I know all the SD card reader needs is an SPI bus and CS line which is exactly like the betaflight target.

mooiweertje, 2021-05-31:
Does this one fix the led strip?

[jnav_2.6.1.4_GEPRCF722.zip](https://github.com/iNavFlight/inav/files/6569408/jnav_2.6.1.4_GEPRCF722.zip)

mooiweertje, 2021-05-31:
The Betaflight config has a DMA config which I never seen in iNav.

mooiweertje, 2021-05-31:
It could be the led strip timer ought to be:

DEF_TIM(TIM1, CH1, PA8, TIM_USE_LED, 1, 0), // LED STRIP D(1, 5, 3)

mooiweertje, 2021-05-31:
If previous build doesn't fix the LED strip this one might.

[jnav_2.6.1.4_GEPRCF722.zip](https://github.com/iNavFlight/inav/files/6569538/jnav_2.6.1.4_GEPRCF722.zip)

mooiweertje, 2021-05-31:
I don't know what's wrong with the SD card reader.

robustini, 2021-05-31:
> 
> 
> If previous build doesn't fix the LED strip this one might.
> 
> [jnav_2.6.1.4_GEPRCF722.zip](https://github.com/iNavFlight/inav/files/6569538/jnav_2.6.1.4_GEPRCF722.zip)

Led now work fine but the pin would like inverted. 
SD Card reader same issue.

mooiweertje, 2021-05-31:
Ok.
I don't know what to do about the SD card. The mapping seems to be all right.

mooiweertje, 2021-05-31:
I updated the previous links to the new build with led strip fixed.

mooiweertje, 2021-05-31:
As far as I know all the SD card reader needs is an SPI bus and CS line which is exactly like the betaflight target.
I don't know what to do about the SD card. The mapping seems to be all right.

robustini, 2021-05-31:
> 
> 
> As far as I know all the SD card reader needs is an SPI bus and CS line which is exactly like the betaflight target.
> I don't know what to do about the SD card. The mapping seems to be all right.

Please can you share the updated target fle? Thanks!

mooiweertje, 2021-05-31:
So all is working so far except the SD card reader?

robustini, 2021-05-31:
Yes!

robustini, 2021-05-31:
> 
> 
> So all is working so far except the SD card reader?

Sorry, I wouldn't want to be wrong, but you declared SDCARD, while this board has an internal flash used as Blackbox storage.
Isn't that the mistake?

robustini, 2021-05-31:
Example, take from "FURYF4OSD" target.h:

// *************** M25P256 flash ********************
#define USE_FLASHFS
#define USE_FLASH_M25P16
#define M25P16_SPI_BUS          BUS_SPI3
#define M25P16_CS_PIN           PB3

robustini, 2021-05-31:
I'm compiling the code with this, I'm confident:

// *************** M25P256 flash ********************
#define USE_FLASHFS
#define USE_FLASH_M25P16
#define M25P16_SPI_BUS          BUS_SPI2
#define M25P16_CS_PIN           PB12

#define ENABLE_BLACKBOX_LOGGING_ON_SDCARD_BY_DEFAULT

mooiweertje, 2021-05-31:
Ah yes, you're right. It's flash not SD.
If your build works please share. If not I will have a look at it again tomorrow using your findings.

> > So all is working so far except the SD card reader?
> 
> Sorry, I wouldn't want to be wrong, but you declared SDCARD, while this board has an internal flash used as Blackbox storage.
> Isn't that the mistake?

robustini, 2021-05-31:
[GEPRCF722.ZIP](https://github.com/iNavFlight/inav/files/6570046/GEPRCF722.ZIP)

[image]

robustini, 2021-05-31:
I had hoped the magnetometer would work, but it didn't, damn.
It finds it but no data moves in SENSOR.

[19 more replies]

closed 2022-02-10 by DzikuVx