iNavFlight/inav #6987: Inav 3.0: weird messages before take off

Mateyhv, 2021-05-18:
Hi,

inicial situation: wing on the ground, gps position locked and acc calibrated.

When awithcing from MANUAL to ACRO and also in the oposite way I got the following messages for a brief second "Motor beeper active" and "Unable to arm" none of which made sense in this situation. Arming as expected was normal regardless of messages. Esc protocol is Dshot. Pics bellow.

Thanks
Matey
[image]
[image]

Jetrell, 2021-05-19:
@Mateyhv Dshot beeper works with all beeper mode if enabled. It effectively takes the place of an onboard buzzer. 
You can disable specific beeper mode items, you do not require.

As for it messaging `motor beeper active`. It is stating the fact. But I understand why its confusing to see it written there.

Jetrell, 2021-05-19:
Dshot beeper works with all beeper modes if enabled `dshot_beeper_enabled = on`. It effectively takes the place of an onboard buzzer. 
You can disable specific beeper mode items you do not require. (like changing modes etc)

As for it messaging `motor beeper active`. It is stating the fact. But I understand why its confusing to see it written there.

breadoven, 2021-05-19:
Seems the messages appear when the beeper is active and you're disarmed because (I assume) you can't arm whilst the Dshot beeper is beeping so it produces messages telling you the Dshot beeper is disabling arming (same as for all the other tings that disable arming).

Seems a bit distracting to me having it flash up every time it beeps when disarmed not helped by the fact it barely appears long enough to read what it says. Perhaps better to remove these messages since there's nothing you can do to re-enable arming other than avoid doing things that cause beeping.

Mateyhv, 2021-05-19:
Thanks @Jetrell and @breadove! I am aware dhsot beeper can be desabled but my point was about the messages flashing on the OSD.

> Seems a bit distracting to me having it flash up every time it beeps when disarmed not helped by the fact it barely appears long >enough to read what it says. Perhaps better to remove these messages [...]

Thats exactly was my thought. I had to toggle through modes several times until I was able to read it. I believe in this situation are irrelevant and quite distracting.

b14ckyy, 2021-05-23:
Got a message from a Pilot that seems to be related: 

>So after flashing to 3.0, I chose wing no tail. I copied my osd and accelerometer settings only and set up everything else manually. Did cli status and it said dshot beeper enabled. Turned it off in cli. Today I went to fly and AutoLaunch went well. After some time i could hear my motor surging. Landed and motor was pretty hot. Got home and checked my cli, dshot beeper was on again. In my video you can see prior to launch system messages displayed unable to arm, motor beeper enabled, but I still managed to launch. I didnt have my goggles down for launch so I didn't know.

Jetrell, 2021-05-24:
@b14ckyy I can't say I've heard motor pulsing from the beeper in flight, via the onboard microphone.
The feature should have an arming flag to prevent this happening. And I would think BLheli_32 Dshot command would as well.

Although, most of my beeper modes are disabled. These are the ones I have enabled.

```
beeper LAUNCH_MODE_LOW_THROTTLE
beeper READY_BEEP
beeper GPS_STATUS
beeper RX_SET
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper HW_FAILURE 
```

But if there is a bug (re-enabling itself). Logs would be the way to track it down.

alccrl, 2021-05-27:
Same issue while using stick commands to load wp mission from eeprom. I had to slow down replay of fpv goggles recording, in order to realize it.
[image]

Alfredo

breadoven, 2021-06-04:
Definitely something odd with these Dshot beeper messages. I'm getting them with standard ESCs, no Dshot used or even available.

avsaase, 2021-06-04:
I had the same problem that I couldn't arm because the dshot beeper was active, with a plane esc that doesn't have dshot. Luckily a friend brought a laptop to the field or wouldn't have been able to fly at all.

Mateyhv, 2021-06-04:
Has never prevented me from arming nor I have noticed anything strange when switching modes inflight (through the 4k cam microphone) I cant tell if the motor is heating more than usual but since I am cruising it usualy comes down cool. Might affect differently depending on individual setup.
Dshot300 on a Holybro Tekko 35A blheli_32 and the FC receives the ESC telemetry without problem.

Mateyhv, 2021-06-05:
Update: previous post was for RC1&2. Yesterday 05/06/21 I built and loaded release_3.0.0 and test it today. Saw the beeper message whenever the Dshot beeper is active (gps fix/etc...) , Armed, no ESC response, test it several times even overriding arming safety, same. Tried autolaunch then armed and same response. Throttle and Autothrottle percent values were correctly updating. Meanwhile all beeps were as expected. Turned battery off and on. Inmediate GPS fix so I armed and ESC worked this time. 
No clue if its directly related to the messages or not. I did about 10 or 15 flights on RC1&2 and never had arming issues before.

breadoven, 2021-06-05:
Doesn't make sense to use the FC defined `USE_DSHOT `as the criteria to decide whether a Dshot ESC is attached to it. Makes more sense to use `motorConfig()->motorPwmProtocol` to define which ESC protocol is actually being used. And the `dshot_beeper_enabled `setting should default to OFF.

alccrl, 2021-06-06:
> Doesn't make sense to use the FC defined `USE_DSHOT`

I do not use DSHOT, I use ONESHOT125 and I get same weird messages, so I looked at the code yesterday, and I ended up in the same reasoning, but since I yet do not know code details, I did not pubblish this idea. So , if I well understood, is this problem caused by a misuse of static @compile time fc features configuration against configurable or @runtime autodetected features ?

Mateyhv, 2021-06-18:
Apparently the problems in this issue are no longer reproducing so I will close it. Thanks everybody, especially the devs!

closed 2021-06-18 by Mateyhv