iNavFlight/inav #6991: 3.0 RTH altitude override issue

Jetrell, 2021-05-19:
@breadoven I tested this feature with a multi rotor and a fixed wing.  It works fine with the quad, but not with the plane.

The mode is enabled `nav_rth_alt_control_override = on` and the `max_check` stick value was exceeded.
I tried it in `EXTRA` and `MAX` altitude modes. And also tried it with `Climb_before_RTH = OFF` and `Spiral_climb` modes. Neither worked with the fixed wing.

The only thing different about the settings between the quad and the plane, that I forgot to test, is `Land_After_RTH`.
The quad is set to ALWAYS and the plane is set to NEVER.

breadoven, 2021-05-19:
This wasn't RTH triggered using switched Failsafe mode by any chance was it ? If it was then the override is disabled during failsafe so wouldn't work ... although it also shouldn't have worked for the multirotor either in that case.

The only difference between MR and FW is the ROLL override doesn't work for a MR but other than that behaviour should be the same when the PITCH override us used.

Can't exactly remember why it was disabled during failsafe, probably just playing safe to avoid an inadvertent override preventing the preset "safe" RTH altitude being reached.

Jetrell, 2021-05-19:
> 
> 
> This wasn't RTH triggered using switched Failsafe mode by any chance was it ? If it was then the override is disabled during failsafe so wouldn't work ... although it also shouldn't have worked for the multirotor either in that case.

@breadoven Thank you. That answers it.
I had a look back through the DVR. That was the case with the plane. I tested it during normal flight. 
While the MR test was done both times in the RTH phase of a waypoint mission,

closed 2021-05-19 by Jetrell

Jetrell, 2021-05-19:
@breadoven I know you think the way I use FS mode, instead of RTH mode is strange.
But there is a reason behind it. And not just to test if FS in the iNav software is working.

For RTH mode switch to work under a failsafe condition. It requires the RX to output a `FS No pulses` or `FS Cut` signal.
Just having the RX failsafe position output a channel to switch the RTH mode to ON, will not work correctly. It won't initiate the FS Landing Procedure if the GPS fix is lost.

I prefer to use RX FS Position though. The reason being, is because I use Pan and Tilt. 
If FS cut is used. The Pan and Tilt servos will not center if a FS occurs. 

That being combined with the previous mentioned. Means the only other option is it use FS mode, with RX FS position to make it all work.
But this now, also limits the way Safehome can be used, as well as Safehome waypoint missions and now the altitude override function too.

breadoven, 2021-05-19:
I have Flysky stuff. It doesn't output any Failsafe signals it just lets you set up the Rx channels to default to certain outputs when it detects failsafe. To use this to trigger failsafe with INAV I set the throttle to go to 800us on the Flysky failsafe. INAV is setup so that any channel input < 900 is considered invalid which triggers INAV failsafe. Once triggered in INAV the INAV failsafe procedure triggers RTH. The Flysky RX failsafe channel setup doesn't do this (trigger RTH)  just the INAV failsafe once triggered by the FLysky Rx throttle channel going out of valid range. Once INAV failsafe is triggered it will act the same as if triggered by any other failsafe signal (no pulses etc). That's the way I've got it setup which should work with any radio that allows you setup specific channel outputs on the RX when the radio gear detects failsafe ... which must be most of the radios out there given they are designed to work in basic models that don't use FCs.

Jetrell, 2021-05-20:
That's interesting. Does Flysky use PPM?
I remember it working the way you stated back in Cleanflight with PPM, when I used Spektrum PPM.

I use TBS Crossfire with OpenTX. Which is outputting CRSF protocol.

When using `FS POS` and setting the RX failsafe throttle position to 750usec. Then returning the throttle back to 1000usec for use. And having iNav `rx_min_usec = 880`.
I can turn off the radio transmitter and nothing happens. The iNav OSD message screen shows nothing. Failsafe is not initiated. Either by RTH with GPS lock or Landing Procedure without GPS lock. 

I can only assume the CRSF FS_CUT outputs a failsafe data packet, that iNav responds to. When the CRSF serial is selected in under the RX tab. And ignores PPM position.   
That is why I can only get CRSF FS_POS to work with FS mode tab selection.
_That would also explain a crash I had 2 years back. I had a hunch that the Crossfire nano locked-up from a noisy pan servo. This left inav clueless. Which lead to the plane slowly taking a dive into the ground._ 

Can I ask. When you get the time. Would you mind looking through the FS code in iNav to get to the bottom of this. 
I don't understand C++. But I do a lot of testing. And generally know what works under most situations with my gear.

Thanks