iNavFlight/inav #6996: Telemetry lost in MATEK F411 Wing using F.port and GPS

Tommy032017, 2021-05-19:
## Current Behavior
Telemetry is lost enabling both GPS and Frsky F.Port over UART1 and UART2 and vice versa.
After disabling GPS under "Configuration" in INAV 2.5.1 / 2.6.1 and also 3.0.0, all values of telemetry works fine again on the radio (OpenTX, Taranis X9 lite and Jumper T18).

## Steps to Reproduce
1.Flash Matek F411 Wing inav_2.5.1_MATEKF411 or newer firmare, even 3.0.0 or newest betaflight
2.RX = FrSky, firmware  RXSR_ACCST_2.1.1_Fport_LBT or 2.1.0
2.Setup ports RX to UART1 (F.port non inverted signal from r-xsr, solder point P) TX1
and GPS (Beitian BN-220) to UART2 or connected GPS / RX to UART vice versa.
[image]
[image]
(set serialrx_halfduplex = ON)

3.Auto search telemetry sensors within newest OpenTX Jumper T18 or FrSky Taranis X9 lite.
4.Sensors are discovered randomly or not at all. Sensors will be recovered only the first 2 seconds after rebooting the FC board F411 Wing.
5. Disabling GPS under "Configuration" in INAV CONFIGURATOR, all telemetry values will get back and refreshed normal, of course except values of GPS.
[image]
[image]

## Expected behavior
All sensors should be recovered continuously with heartbeat.  

## Suggested solution(s)
There must be an overlay or mismatch in telemetry handling in the INAV firmware for F411 Wing. It doesn't matter which firmware is used (INAV or even Betaflight). The wrong coding must be in telemetry handling of F.port, because using FrSky inverted S.port and SBUS within soft serial works under the same hardware without any problem.

## Used hard- / software and versions
INAV/MATEKF411 3.0.0 May 15 2021 / 16:59:12 (7b6a37a3)
GCC-9.3.1 20200408 (release)
FrSky R-XSR,  RXSR_ACCST_2.1.1_Fport_LBT or 2.1.0
OpenTX, Version 2.3.11-otx, Jan 8 2021
Jumper T18, opentx-t18-lua-noheli-de-2.3.11-otx.bin

FLIGHT CONTROLLER F411-WING
STM32F411, MPU6000, INAVOSD, BMP280, 2x UARTs, 1x Softserial, 1x I2C, 2x Motors & 5x Servos, 2x BEC & current sensor on board.

GPS: Beitian BN-220

Mateyhv, 2021-05-19:
Your setup seems correct. My main concern will be the RXSR with the new ACCST 2.1.1, can you try with a different reciever?

Tommy032017, 2021-05-19:
Yes, your concern could be most likely!
First I ordered the FrSky Archer RS EU LBT  and I mean it was working fine, but I don't remember for sure, because I tested it only for a short moment. I sent the receivers back, because the Jumper T18 only has ACCST D16 within the JP5in1 module and does not work with ACCESS and Archer RS.
I also tried different firmware for the R-XSR:
RXSR_ACCST_2.1.1_Fport_LBT and RXSR-**FC**_ACCST_2.1.1_LBT (OMNIBUS F4),
without any difference and same telemetry problem.
I do not have any other receiver with F.port and ACCST for testing.

Mateyhv, 2021-05-19:
I am on Fport LBT on R9slim+ reciever ACCST and have had no problems with telemetry so far even on Inav 3.0rc1. Bear in mind this is ACCST v1 and Fport v1 as well, but thats the only real difference  as everything else in your setup looks good. Maybe someone using the v2 can chime in.

Tommy032017, 2021-05-20:
I sent a request to FRSky CN and they will do some tests and hopefully get back with a solution.
@Mateyhv: Thank you for your help.
I also tested the Archer RS switched to protocol F.port 2 and set INAV 2.6.1 (serial Fport 2) and it didn't work all, even lost of all channels and stick movements.

MRC3742, 2021-05-21:
After spending hours researching this same problem I found some open issues on BF and FrSky github pages.
R-XSR firmware update 2.1.1 seems to be causing telemetry loss.
I posted the findings in Discussion https://github.com/iNavFlight/inav/discussions/6607 with a link to the open issue on FrSkyRC/Firmware-Test

Reverting the R-XSR firmware back to version 2.1.0 fixed my no telemetry issue after enabling GPS !!!

Rich

Tommy032017, 2021-05-21:
Thanks to @Mateyhv and @MRC3742 with their right guess and temporary solution!
I also spent hours and days with this problem. Opening the issue again was finally a good idea.

**The problem seems now to be very clearly by FrSky and the latest firmware RXSR_ACCST_2.1.1_Fport_xxx.**

I just flashed back to version **RXSR_ACCST_2.1.0_Fport_LBT.frk and it fixed the problem** with no telemetry too, using RX and GPS!
I'll post more information here, after FrSky will get back with more detailed information and solution to their actual firmware 2.1.1.

MRC3742, 2021-05-22:
FrSky has posted a beta firmware for testing.
https://github.com/FrSkyRC/Firmware-Test/files/6520836/RXSR-RXSR-FC-fport-beta-0521.zip

Tommy032017, 2021-05-22:
@MRC3742 Tank you very much. 
I tested RXSR_Accst_Fport_LBT_2.1.2-beta.frsk and everything seems to work perfect - no more issue using RX + GPS and telemetry lost. Real flight still pending.

pilotltd, 2021-06-23:
Similar issue with Matek 405 STD - limited telemetry available.