iNavFlight/inav #6997: [3.0-RC1] WP numbering issue

Mateyhv, 2021-05-19:
-After a multile jumps in sequence after the same waypoint, the numbering sequence fails.

Eg; wp12coord, wp13jump (hidden), wp14jump (hidden), wp15coord. So wp12 is numbered 12 and wp15 should be 13, however when there is a double or multiple consecutive jumps wp15 is shown as 15 and the numbering of all following jumps even single jumps fail skipping one number for the jump itself. Mission Control handles the numbering correctly.

-On two digit wp number only the unit number is show (I guess thats know, see pic bellow)

-The message "PREPARING FOR NEXT WAYPOINT" flashes sometimes so fast its just one or two frames of the DVR. Not sure whats the meaning, I guess the FC is doing some calculations before switching to the next WP.

[image]

breadoven, 2021-05-19:
https://github.com/iNavFlight/inav/pull/6429 fixes the inconsistent WP numbering issue caused by non geo spatial WPs such as Jump. Still not merged.

With the HUD WP number showing 1 digit rather than 2  ... it's as intended rather than a bug. Something to do with a lack of room to display > 1 digit.

Preparing for next WP ... shows when the next waypoint is being set up. Probably another message that's not much use given it barely has time to display given how quickly the next WP setup is processed.

Mateyhv, 2021-05-19:
Thanks @breadoven, I supposed the wp second digit was not a bug, hardly to go unnoticed. The "preparing.." message is totaly impossible to see, during the flight I only noticed kind of a glitch a couple of times. Only knew what happened reviewing the DVR and its not more than two frames but yes you are left wondering what had happened during the flight :)

So if you think everything is covered including the "Raise" typo that I just added I am going to close the issue so it gets out of the way.

breadoven, 2021-06-07:
@Mateyhv #6429 was merged which should fix this issue.

Mateyhv, 2021-06-07:
Thank you @breadoven! Closing

closed 2021-06-07 by Mateyhv