iNavFlight/inav #7005: CMS configurable 'nav_wp_load_on_boot'

Mateyhv, 2021-05-20:
When the new setting ```nav_wp_load_on_boot=ON``` and the pilot do not erase the waypoints from eprom before going to the next flying location which is greater than ```nav_wp_safe_distance``` that will prevent arming and efectively ground the model as none of the settings are configurable without connection to Configurator/Speedybee.

Possible solutions:
CMS configurable ```nav_wp_load_on_boot``` setting.
CMS option to erase mission from FC volatile memory.
CMS option to erase mission from eprom memory.
CMS configurable ```nav_wp_safe_distance``` if 0 means unlimited.

Thanks

Jetrell, 2021-05-20:
@Mateyhv 
There is already function in this feature to unload the mission.  Just use the stick position that loads the mission.. and it will unload it.

`I've also changed the stick command for loading waypoints to be like a toggle. It unloads waypoints if waypoints are loaded. This is needed if the waypoint mission's first waypoint is too far from the arming point. When it is too far, it will prevent you from arming.`

The interesting part. Is that I tested it with a mission loaded 12km away. Mine still armed, even with the distant mission loaded.

pbreed, 2021-05-21:
Really don't like the WP) load toggle, because you can'ty be sure the state of the world.
should be SEPERATE stick positions for LOAD WP  and UNLOAD.

IE did I run this already or not? How shal I know?

Jetrell, 2021-05-21:
> 
> 
> Really don't like the WP load toggle, because you can't be sure the state of the world.
> should there be SEPERATE stick positions for LOAD WP and UNLOAD.
> 
> IE did I run this already or not? How shall I know?

@pbreed If it won't arm due to a waypoint mission being `auto loaded` with another location. 
Disarm the craft. Then move the sticks to the position that would load a mission. 
Then try to arm again. If it arms, then its all good.  
You will know it unloaded the mission, if you try to start a mission with the WP mode switch...Nothing will happen.

Other than that, you can turn the feature off. Or load a new mission with a PC of smartphone at the location.

breadoven, 2021-05-21:
There's a new OSD Mission field in https://github.com/iNavFlight/inav/pull/6967 that shows the number of waypoints loaded. There's also a message on the Arming screen telling you if a mission is loaded, although if you don't get as far as the arming screen because arming is blocked by the fact the first waypoint is too far then that's obviously telling you in itself a mission is loaded.

Perhaps a lot of these new option just need explaining better in the "documentation". Most of them are only briefly covered in the CLI settings document.

And it's not possible to erase a mission from the EEPROM, only overwrite a previously stored mission. WP reset only flags a mission as invalid and sets the waypoint count to 0 but it doesn't actually delete the WP entries in the EEPROM or even in FC volatile memory for that matter.

Mateyhv, 2021-05-21:
Thanks everybody for answering! I have updated the original post. Yes, documentation is needed because without tracking the original PR from @tonyyng the impossibility to arm might catch a few by surprise. 

I still believe the option to change ```nav_wp_load_on_boot``` through CMS (maybe in the fixed wing, navigation folder) is a good addition so the pilot does not have to autoload and manualy unload the mission each time when Configuratos is unavailable.

Thanks

tonyyng, 2021-05-21:
I'm not sure anyone is really caught by surprise.  The default value for nav_wp_load_on_boot is OFF.  You have to explicitly turn it on to have it load the mission from EEPROM.

I'll have a look at adding the option to the CMS menu.

pbreed, 2021-05-22:
Maybe I don't understand something...
I'm using DJI /Caddax vista so my osd screen display is limited.
I take off manually... then engage WP mode in flight....
If the stick position toggles the WP in/out how do I know the WP's are loaded before I take off?
My issue is not with the nav_wp_load_on_boot = ON
My concern is the statement:

I've also changed the stick command for loading waypoints to be like a toggle. It unloads waypoints if waypoints are loaded. This is needed if the waypoint mission's first waypoint is too far from the arming point. When it is too far, it will prevent you from arming.

This completely breaks my way of using this...
With 2.6.1 as long as sometime in the process I've put the sticks in the load position I can be sure the WP's are loaded...
Now I can't be sure..

Did I misunderstand this?

tonyyng, 2021-05-22:
Here's your new way of thinking about this: (translating your statement)

If I have nav_wp_load_on_boot = ON, and a mission stored in EEPROM,  I can be sure that the WP's are loaded....

Don't use the stick command and you won't be confused about the state of the world.  The stick command is only needed for the scenario where you try to arm, but it won't let you because the loaded mission has a first WP that is too far away.  If you are in this scenario, you'll know as well, since you won't be able to arm (or fly).

FPVZaphod, 2021-05-24:
Wouldn't it be better and easier to always load the mission on boot ( if nav_wp_load_on_boot = ON ), but only accept engaging Waypoint Flightmode if the Distance to the 1st Waypoint it below treshold? This would both give safety and ease of usage. Perfect would be an Error Message in the OSD, stating "Waypoint too far" when enabling WP Mode too far away :-)

tonyyng, 2021-05-24:
> Wouldn't it be better and easier to always load the mission on boot ( if nav_wp_load_on_boot = ON ), but only accept engaging Waypoint Flightmode if the Distance to the 1st Waypoint it below [...]

That's exactly what I've done in my own personal version of iNav.  I'll think about adding it in the next release.  There are other WP related changes being added by others.  I'd like to see what those are before adding my changes.

Mateyhv, 2021-05-24:
Indeed thats a much better for the new logic. Before it was the pilot who loaded the mission because he wishes to fly it so the system has the inbuilt security to prevent arming if the Flight Plan was mistaken. So he only has to cycle power to boot without a FP Now if the FP is always loaded there is no reason to prevent arming but instead not to engage WP mode with the corresponding warning.

@tonyyng , given that you provided a way to unload the FP now (my fault for not reading the full PR) and the possible future change in FP managment I am going to close this issue. Thanks!

closed 2021-05-24 by Mateyhv

Jetrell, 2021-06-26:
> That's exactly what I've done in my own personal version of iNav. I'll think about adding it in the next release. There are other WP related changes being added by others. I'd like to see what those [...]

@tonyyng Any chance of adding the changes you spoke of here? Now that a fix is needed in the Configurator mission planner?

stronnag, 2021-06-26:
The configurator does not need fixing, the firmware does (#7190)