iNavFlight/inav #7012: Target request RDQ JB F7 flight controller

petertorelli, 2021-05-22:
## Current Behavior
iNav does not suppirt my flight controller: [RDQ JB F7](https://www.racedayquads.com/products/rdq-bardwell-f7-3-6s-30x30-flight-controller-jbf7-fc) 

## Desired Behavior
iNav supports my flight controller.

## Suggested Solution
Since Clearflight and Betalfight both support it, and it is an F7, this shouldn't be too far off from already is supported? (STM32F7)

## Who does this impact? Who is this for?
This impacts anyone using my brand of FC.

## Additional context

petertorelli, 2021-08-16:
JBF7 seems too suited to racing, not iNav. Closing.

closed 2021-08-16 by petertorelli

Cimos, 2021-09-19:
I am keen on getting iNav working on my RDQ JB F7. Main reason is Ive got a spare one i am wanting to use. I realize this FC might be somewhat limited in the iNav world as it was primary designed for racing. 

I am happy to do the leg work. Can someone point me in the right direction?