iNavFlight/inav #7013: 3.0 RC1 Erroneous values in the blackbox log, for distance, azimuth and mah/km

OlivierC-FR, 2021-05-23:
Distance to aircraft is wrong by few millions of meters, azimuth stays at 7 for the whole flight, and mah/km is some seemingly random values. After take of, mah/km is back to normal, but the 2 others stays very wrong for the whole flight.

[LOG10005.01.csv](https://github.com/iNavFlight/inav/files/6527784/LOG10005.01.csv)

[image]

stronnag, 2021-05-23:
Official blackbox_decode does not show these derived values.
Some older RCs did, but the feature was removed for precisely the reasons shown in your image.

stronnag, 2021-05-23:
This is not a firmware issue,  see https://github.com/iNavFlight/blackbox-tools/pull/30 and linked PRs etc.

OlivierC-FR, 2021-05-23:
Ok thanks, will check blackbox version, can close the PR then.

stronnag, 2021-05-23:
Apologies, on examination of **BBL** it is a firmware issue, the home position is no longer being recorded on the log.

Possibly as a result of the new safehome processing (@tonyyng  any comment?)

OlivierC-FR, 2021-05-23:
Yep home point is definately at 0N 0E

[image]

stronnag, 2021-05-23:
As is shown by the **stable release** `blackbox_decode`:

[image]

tonyyng, 2021-05-23:
> Possibly as a result of the new safehome processing (@tonyyng any comment?)

I don't use the blackbox logs with my planes since my flight controllers don't have enough memory to use it.  I tried looking at the code, but it is over my head.

@OlivierC-FR Do you have any safehomes defined?  Try a flight without any safehomes defined.  If the problem goes away, then the problem is likely related to safehomes.  If the problem still occurs, then it is likely not safehomes.

stronnag, 2021-05-23:
Don't think it's to do with safehome, let me look further.

stronnag, 2021-05-23:
On my quad, when armed, home is written to the BBL (it did have safehome enabled too)

@OlivierC-FR can you post your `diff` here please, let's try and see what ninja settings are causing this.

OlivierC-FR, 2021-05-23:
https://www.dropbox.com/s/m6u2sj76km7gsjy/Q5-F722SE.diff.txt?dl=0
https://www.dropbox.com/s/epd20sutrbc177i/Q5-F722SE.dump.txt?dl=0
No safehomes.

stronnag, 2021-05-23:
Thanks. So far I'm somewhat stumped on this. Works fine here on one quad regardless of safehome.

wx4cb, 2021-05-23:
here's a couple of log files from the other day, I don't have BB Explorer on this machine, but maybe it's useful to you.

No safe homes set

[LOG00014.TXT](https://github.com/iNavFlight/inav/files/6528949/LOG00014.TXT)
[LOG00007.TXT](https://github.com/iNavFlight/inav/files/6528950/LOG00007.TXT)

stronnag, 2021-05-24:
The mystery deepens. @OlivierC-FR 's is the only log I've seen where this happens. It's only the only log I've seen recently where BBL is on a switch rather than auto on ARM.

stronnag, 2021-05-24:
Sheesh !
```
set nav_extra_arming_safety = OFF
```
All bets are off. You armed without a GPS fix (and no home).

OlivierC-FR, 2021-05-24:
I went to the DVR and it does not tell much, I pressed the record button right after arming. It says 15 sats, but no way to be sure there was a GPS fix couple of seconds before. Anyway, in the case of no GPS fix, shouldn't it record as home the 1st correct GPS coordinates ? Or is it too late ? Like, if it hasn't happened at arming, no way to do it later I guess ?

[image]

stronnag, 2021-05-24:
> yway, in the case of no GPS fix, shouldn't it record as home the 1st correct GPS coordinates ? Or is it too late ? Like, if it hasn't happened at arming, no way to do it later I guess ?

You'd have hoped so. But it appears not, I may look a bit deeper, but this is no longer an "oh sh*t" issue for 3.0 IMO.

OlivierC-FR, 2021-05-24:
Generally speaking I do not take-off without a GPS fix, but in this particular case maybe I didn't wait for the fix, I don't recall. What's for sure I had zero plans to fly further away that few hundreds of meters. So let's say its probable the lack of GPS fix at arming caused that.

OlivierC-FR, 2021-05-24:
And what about this one ? : https://github.com/iNavFlight/inav/issues/6999
Incorrect mah ? Could it be somehow related ?
This one I for sure did not throw a plane without a proper GPS fix.
100%

stronnag, 2021-05-26:
The code is quite clear; home is only set when (a) there is an adequate GPS fix and (b) the craft is not armed.
The blackbox behaviour is entirely consistent in this case.
So let's close this.

closed 2021-05-26 by stronnag