iNavFlight/inav #7015: mission WP speed always around 10 km/h in OSD, not as described in mission file and uploaded to EEPROM

alccrl, 2021-05-23:
This eeprom mission file : <?xml version="1.0" encoding="UTF-8" standalone="yes"?>
<mission>
	<version value="2.3-pre8"/>
	<mwp cx="7.5398397" cy="45.0423113" zoom="17"/>
	<missionitem no="1" action="WAYPOINT" lat="45.0424495" lon="7.5396223" alt="20" parameter1="1000" parameter2="0" parameter3="0"/>
	<missionitem no="2" action="WAYPOINT" lat="45.0418654" lon="7.5383517" alt="40" parameter1="2000" parameter2="0" parameter3="0"/>
	<missionitem no="3" action="WAYPOINT" lat="45.0402911" lon="7.539917" alt="60" parameter1="3000" parameter2="0" parameter3="0"/>
	<missionitem no="4" action="POSHOLD_TIME" lat="45.0415055" lon="7.5420492" alt="40" parameter1="20" parameter2="0" parameter3="0"/>
	<missionitem no="5" action="LAND" lat="45.042181" lon="7.5396072" alt="20" parameter1="0" parameter2="0" parameter3="0"/>
</mission>

is executed at 10 Km / h gps speed, not as described in mission file .

Environment:
Mateksys 722-STD, gps, mag, baro, NAV_POS_HOLD while starting Wp mission MODE of a custom built QAV-250 quad.
Mission is executed as expected (included hold for 20s on wp4) , except for quad speed, that is always shown around 10 km/h, for every wp leg 1,2,3,4. I tried to look at black box files, but it is not clear the mapping of the various speeds.

Alfredo

Any idea ?

stronnag, 2021-05-23:
Proabably, you have set the `nav_auto_speed` in the cli, the mission will not exceed that.

alccrl, 2021-05-23:
I just checked : I have set nav_auto_speed = 300 in my configuration.

stronnag, 2021-05-23:
So working as designed then.

closed 2021-05-23 by stronnag

alccrl, 2021-05-23:
Thanks,
but I would have close the issue after a retest and check that documentation is enough clear about it, in other words after a check about what design is, because settings.md and inav configurator do not specify exactly what the setting nav_auto_speed is for.
I work quite always In POS_HOLD (that is part of nav), and I can reach 100 km/h in such situation, no limits lay around.
Some time the prefix "nav_" may be confusing. In this case must be associated to waypoint nav, that is a nav subset.

Am I wrong ?

Alfredo