iNavFlight/inav #7016: 3.0 RC1 - Blackbox logging rate not able to be set higher than 25% (1/4) on F405 wing

wx4cb, 2021-05-23:
F405 Wing, Latest release candidate, not sure if this is a configurator issue or not, but you cannot set the logging rate higher than 1/4 (25%). 

Symptom: Logging rate shows as 25 %

To Reproduce: Set it higher than 25% (i usually do 100%)

let it reboot and go back in to the BB tab and it's back at 25% again

```
# get black
blackbox_rate_num = 1
Allowed range: 1 - 65535

blackbox_rate_denom = 4
Allowed range: 1 - 65535

blackbox_device = SDCARD
Allowed values: SERIAL, SPIFLASH, SDCARD

```

[image]

avsaase, 2021-05-23:
Do you use 4kHz loop time? I ask because blackbox logging is limited to 1kHz.

stronnag, 2021-05-23:
#6127 probably

wx4cb, 2021-05-23:
> Do you use 4kHz loop time? I ask because blackbox logging is limited to 1kHz.

that's entirely possible, lemme check in a minute - never thought about that

wx4cb, 2021-05-23:
> #6127 probably

most likely.

wx4cb, 2021-05-23:
yea thats what it is....

closed 2021-05-23 by wx4cb