iNavFlight/inav #7033: Better Navigation by combining GPS direction and compass

b14ckyy, 2021-05-27:
I have seen the feature request here #6650 that looks interesting too but I have an idea of a different approach. 

My idea is to have a Compass Heading / GPS direction transition, based on the deviation between both. 
We all know that our compasses are not the most reliable and precise ones and on some planes it is harder to place the compass away from power wires than on some quads. So using the GPS heading as the main navigation source makes still sense. But thus causes trouble in strong headwinds. 

My Idea is to have specific "zones" where just GPS heading is used or a transition to compass heading will happen. 

Example: We fly in no wind or just slight wind conditions. The plane is in RTH or waypoint mode and aims at the target point. It compensates for crosswind because of the GPS heading used and the deviation between plane Heading and Ground Course is less than (lets say) 20° and all is fine. Perfect course. 

Now the wind gets stronger. To keep the course the plane turns it's nose against the crosswind more to stay on ground course but ground course slows down. Now the nose points 30° against the wind and the Ground Course and Compass reading is also 30° off. This would be inside a transition zone that might be in between 20° and 40°. So 50% weight on the compass and 50% of the GPS ground course is used. This will now shift the navigation course to 15% off target and the plane will turn back to get the course on target again. 

[image]

This should result in a soft bearing limitation, so the course can never be off the plane heading by more than 40°. If the craft has now a slow airspeed compared to windspeed, the Minimum ground speed safety will take over and throttle up. This might still cause some snaking in extreme conditions but will still 100% ensure that the plane will neer loose course and come back home.

There is also another feature request that will base on this one.

wx4cb, 2021-05-27:
but the course over ground aka the track (which is the important one as regards to RTH) should still be whatever direction the plane needs to be going in order to get home irrespective of the heading that the nose is pointing at (which is where a compass would be handy).

I found this in a quick search, maybe there's a way we can do this simple math? 

[https://aviation.stackexchange.com/questions/3871/what-is-a-general-rule-for-crosswind-correction-especially-when-landing](url) 

```
What I have learned during my commercial flight training is making use of the speed number. Take your TAS, divide it by 60. This is your speed number.

Now take you XWC (crosswind component). Divide the XWC by your speed number. This is the amount of degrees you should crab to stay on track (wind correction angle)

Lets use an example:

We are flying in a C172 at 120kts TAS. XWC is 18kts from the left. 120 divided by 60 is 2, so our speed number is 2. 18kts wind divided by 2 is 9. Now adjust your heading by 9 degrees to the left (into the wind), and you should stay on track.

Worked perfectly fine for me so far. Hope it helps! Cheers

```

rts18, 2021-05-27:
> 
> 
>
> We all know that our compasses are not the most reliable and precise ones and on some planes it is **harder to place the compass away from power wires than on some quads**. So using the GPS heading as the main navigation source makes still sense. 

Don't under estimate mag interference on quads.

#6206 Is based on multi-rotor use. 

When a fixed wing requires a magnetometer for navigation; It generally indicates that the user set `nav_fw_cruise_thr ` too low. 
A magnetometer may keep the plane facing in the correct direction in a strong headwind. But it won't get it back home in failsafe, if the throttle settings were based on efficiency. rather than accounting for poor atmospheric conditions.

b14ckyy, 2021-05-28:
> but the course over ground aka the track (which is the important one as regards to RTH) should still be whatever direction the plane needs to be going in order to get home irrespective of the heading that the nose is pointing at (which is where a compass would be handy).

That's exactly what I describe above.  it is good to keep the course over ground as the main navigation reference but in some cases with strong headwinds, it would be avoided to point the nose too far away from the target. Like in crosswinds. In this case the compass should take over and maybe also warn the pilot about strong winds or compass Deviation. 

>When a fixed wing requires a magnetometer for navigation; It generally indicates that the user set nav_fw_cruise_thr too low.

No. That just means that the wind is stronger than the Cruise Airspeed. No long range plane is tuned for high speeds and wind speed can always suddenly go up. 
Also INAV has a integrated minimum ground speed. If the speed goes too low, it will automatically throttle up to make it home. But it needs the right direction. And the heading limitation can ensure this.

rts18, 2021-05-29:
> No. That just means that the wind is stronger than the Cruise Airspeed. No long range plane is tuned for high speeds and wind speed can always suddenly go up.
> Also INAV has a integrated minimum [...]

I understand your perspective. And don't have an issue with using a mag to augment fixed wing GPS heading.
Although, I have observed that many pilots tune `nav_fw_max_thr` far too low. e.g. 1600uSec.
Remembering that _minimum ground speed throttle increase_ is constrained by the `nav_fw_max_thr` value. 

  if ((navStateFlags &amp; NAV_CTL_ALT) || (navStateFlags &amp; NAV_CTL_POS)) {
      
      
                uint16_t correctedThrottleValue = constrain(posControl.navConfig-&gt;fw_cruise_throttle + throttleCorrection, posControl.navConfig-&gt;fw_min_throttle, posControl.navConfig-&gt;fw_max_throttle);
      
      
                rcCommand[THROTTLE] = constrain(correctedThrottleValue, posControl.escAndServoConfig-&gt;minthrottle, posControl.escAndServoConfig-&gt;maxthrottle);
      
      
            }

If a plane has its nav throttle values tuned correctly, 25kph(7m/s) is plenty of forward speed to maintain a GPS heading before `nav_fw_cruise_thr` will potentially increase to the `nav_fw_max_thr` value, and prevent the plane being stopped or pushed backwards.

b14ckyy, 2021-05-29:
that's a good point. Do you really thing many pilots change the max throttle? 
I usually set it to 2000 unless there are limitations for the motor or battery that could get damaged on full throttle. But this only applies for high performance setups but not my long range cruisers.

rts18, 2021-05-29:
> 
> 
> that's a good point. Do you really thing many pilots change the max throttle?
> I usually set it to 2000 unless there are limitations for the motor or battery that could get damaged on full throttle. But this only applies for high performance setups but not my long range cruisers.

From years of looking over FC settings and helping people with crashes etc. Not just with iNav, but other flight software as well. I would say Yes. Many people reduce _nav max throttle_, trying to make the plane more efficient. 
And they do so on long range builds. Which is the place that they should not reduce it. The reason being. Its impossible to say what the atmospheric conditions may be like many kilometers away from takeoff and especially in mountainous regions. 

I've looked over many DVR's and logs from crashes. 
It was easy to see that the FC knew what it should be doing. But lacked the thrust(air speed) to accomplish it under the conditions. 

But I do believe the addition of a mag for fixed wings would be helpful if a GPS heading is lost for a short time, with the ability to account for cross-track error.

wx4cb, 2021-06-06:
@b14ckyy marc, yes they do,. there was a dvr post of exactly that in the IFWG just the other day from memory.

I personally am like you, i set my max to 1990 cus i want all available go around power for this exact reason, but my cruise throttle is extremely low on the light builds (like the drift) on most other planes it's at least 1450

b14ckyy, 2024-03-20:
pretty much covered with all the AHRS and Nav system changes since 6.0

closed 2024-03-20 by b14ckyy