iNavFlight/inav #7036: COMPASS ERROR

Minecraftman04, 2021-05-28:
## Current Behavior
Mag is slow after loading. https://youtu.be/rDArl-Z9wI0 (It was sunny) This causes it to be unusable in flight. Problem is on multiple BN880's and different boards and firmware, 
## Steps to Reproduce

1. Have mag in BN880
2. Configure correctly
3. Wait
4.

## Expected behavior
Smooth movement. 

## Suggested solution(s)
No idea
## Additional context
# dump

# version
# INAV/MATEKF722WPX 3.0.0 May 15 2021 / 17:08:22 (7b6a37a3)
# GCC-9.3.1 20200408 (release)

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 0 50 0 -1
smix 1 3 1 50 0 -1
smix 2 4 0 -50 0 -1
smix 3 4 1 50 0 -1
smix 4 5 6 100 0 -1

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# safehome
safehome 0 0 0 0
safehome 1 0 0 0
safehome 2 0 0 0
safehome 3 0 0 0
safehome 4 0 0 0
safehome 5 0 0 0
safehome 6 0 0 0
safehome 7 0 0 0

# logic
[... 1401 lines cut ...]
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 425
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

# end the command batch
batch end
---

- FC Board name and vendor: Matek f722 wpx / Omnibus f4v3
- INAV version string: 2.6.1 or 3.0.0 RC1

b14ckyy, 2021-05-31:
this is normal. The mag is used for attenuating to the correct heading. The Realtime yaw positioning is done via gyro. Like the Accellerometer that is used to detect where the ground is but only slowly attenuates. 

Install it to the frame and then it will be fine.

Minecraftman04, 2021-06-01:
> this is normal. The mag is used for attenuating to the correct heading. The Realtime yaw positioning is done via gyro. Like the Accellerometer that is used to detect where the ground is but only slowly attenuates.
> 
> Install it to the frame and then it will be fine.

ah thank you

closed 2021-06-01 by Minecraftman04