iNavFlight/inav #7037: Mavlink issue in 3.0

mtnmikev, 2021-05-28:
## Current Behavior
Mavlink on UART5 of the Matek F405 wing FC wired to a TBS Nano RX sending telemetry via Bluetooth to MFD tracker.
Working on  Inav 2.6 and not working on 3.0

## Steps to Reproduce

1.Reflash to 2.6 with same settings as 3.0
2.Confirm working or not. 
3.
4.

## Expected behavior
100% link to tracker
## Suggested solution(s)
Don't know
## Additional context

3.0 dump

# version
# INAV/MATEKF405SE 3.0.0 May 15 2021 / 16:56:35 (7b6a37a3)
# GCC-9.3.1 20200408 (release)

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 0 50 0 -1
smix 1 3 1 50 0 -1
smix 2 4 0 -50 0 -1
smix 3 4 1 50 0 -1

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1461 100
servo 4 1000 2000 1586 -100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# safehome
safehome 0 0 0 0
safehome 1 0 0 0
safehome 2 0 0 0
safehome 3 0 0 0
safehome 4 0 0 0
safehome 5 0 0 0
safehome 6 0 0 0
safehome 7 0 0 0

[... 2759 lines cut ...]
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

# end the command batch
batch end

---

see above
- FC Board name and vendor:
- INAV version string:

kasatka60, 2021-05-30:
matek f722-wing inav 3.0RC and 3.0RC2. Mavlink  not working.
RC: qczek rls 2.10

stronnag, 2021-05-30:
I rather fear that mavlink v1 is broken in 3.0; from some small bench tests and a legacy data capture

* All inav **v3** Mavlink v1 CRCs are incorrect
* Mavlink v2 CRCs are correct, is the capture from legacy firmware.

Sample samples in the attached archive:
* mav-2018-02.raw : captured from inav / CRSF 3 years ago
* m1-2021-05.raw : captured from inav 3 today, set to mavlink V1
* m2-2021-05.raw  : captured from inav 3 today, set to mavlink V2
* mavreader.go : simple little MAV CRC checker.

```
# inav 3, Mavlink V1 :(
$ ./mavreader m1-2021-05.raw
OK 0, fail 725
# inav 3, Mavink V2 :)
$ ./mavreader m2-2021-05.raw
OK 682, fail 0
# inav 2.?, Mavink V1 :) OTA CRSF, expect a few bad packets
$ ./mavreader mav-2018-02.raw
OK 102, fail 23
```
[mavtest.tar.gz](https://github.com/iNavFlight/inav/files/6566455/mavtest.tar.gz)

@harry1453 any comments. Does v1 work for you?

harry1453, 2021-05-30:
@stronnag So you're saying MAVLink v2 works but v1 doesn't? Let me investigate.

stronnag, 2021-05-30:
> @stronnag So you're saying MAVLink v2 works but v1 doesn't? Let me investigate.

That's what it looks like to me. I have a sad history of investigating mavlink crc issues ...

wx4cb, 2021-05-30:
> > @stronnag So you're saying MAVLink v2 works but v1 doesn't? Let me investigate.
> 
> That's what it looks like to me. I have a sad history of investigating mavlink crc issues ...

ok come on @stronnag - You've never steered me wrong :D

mtnmikev, 2021-05-30:
I switched to V1 and V2 in CLI.......no dice.

stronnag, 2021-05-31:
@harry1453 just FYI, I reflashed the bench-board with 2.6.1 and the CRC is perfect again, so this is definitely the issue.

mavlink 1.0, inav 2.6.1
```
Mav1: 74 20 : vfr hud a: 0.9 g: 0.0 h: 350 thr: 0 a: 0.2 cl: 0.0
Mav1: 0 9 : Heartbeat: t: 1 a: 0 b: 209 s: 4 m: 3
Mav1: 35 22 : RC Chan t: 101871 1: 1497 2: 1500 3: 1502 4: 998 r: 0
Mav1: 30 28 : Attitude: t: 101891 r: -0.2 p: -0.0 y: 6.1
V1 OK 854, fail 0
```
As an aside, all the data looks plausible (my trivial validator only checks a subset) for the inav 3.0 / Mavlink 2.0 set as well.

harry1453, 2021-05-31:
Okay, looks like the issue is *likely* that INAV 3.0 isn't switching the library into MAVLink v1 okay.

The MAVLink [documentation](https://mavlink.io/en/mavgen_c/#sending-and-receiving-mavlink-1-packets) expects you to use the new channel feature in the new MAVLink library which INAV doesn't use. INAV probably isn't setting the flag correctly.

Here's where it's done in INAV: https://github.com/iNavFlight/inav/blob/c47c6edb56da45ee0763dcf14d4d23b6c727dea0/src/main/telemetry/mavlink.c#L329

Maybe we should implement version handshaking whilst we're at it...

stronnag, 2021-05-31:
yeah, that does the trick. Anything else is a consumer problem. 

```
Mav1: 0 9 : Heartbeat: t: 1 a: 0 b: 209 s: 4 m: 3
Mav1: 30 28 : Attitude: t: 95915 r: -0.1 p: -0.0 y: -0.1
Mav1: 147 36 : Bat Status: c: 0 v0-4: 3510 3510 3510 -1
Mav1: 29 14 : Pressure: t: 95975 0.0 t°: 5490
Mav1: 30 28 : Attitude: t: 96036 r: -0.1 p: -0.0 y: -0.1
Mav1: 35 22 : RC Chan t: 96097 1: 1498 2: 1500 3: 1503 4: 998 r: 0
Mav1: 30 28 : Attitude: t: 96158 r: -0.1 p: -0.0 y: -0.1
V1 OK 1542, fail 0
V2 OK 1, fail 0
```
When you're done, if you'd rebase against the `release-3.0.0` branch, I'll merge it.

harry1453, 2021-05-31:
Done!

stronnag, 2021-05-31:
Seems like it sends a V2 stanza initially, then what is set. Can't see that being an issue for sane consumers.
```
$ ./mavreader /tmp/x1.raw 
Mav2: 1 31 : Status: l: 50 v: 9720 c: 0
Mav1: 35 22 : RC Chan t: 120609 1: 1499 2: 1498 3: 1501 4: 998 r: 0
...
Mav1: 1 31 : Status: l: 60 v: 9640 c: 0
Mav1: 74 20 : vfr hud a: 0.0 g: 0.0 h: 355 thr: 0 a: 0.0 cl: -0.0
Mav1: 0 9 : Heartbeat: t: 1 a: 0 b: 209 s: 4 m: 3
Mav1: 35 22 : RC Chan t: 140664 1: 1499 2: 1498 3: 1501 4: 998 r: 0
Mav1: 30 28 : Attitude: t: 140664 r: -0.1 p: -0.0 y: -0.1
Mav1: 30 28 : Attitude: t: 140786 r: -0.1 p: -0.0 y: -0.1
V1 OK 414, fail 0
V2 OK 1, fail 0

```

harry1453, 2021-05-31:
That's pretty odd, no idea why it would do that as we set the flag before sending any message.

If that's okay, I think this can be closed now

stronnag, 2021-05-31:
Closing as fix #7047 merged.

closed 2021-05-31 by stronnag

mtnmikev, 2021-05-31:
Hey guys, Thanks!  How do I test this?

stronnag, 2021-05-31:
> Hey guys, Thanks! How do I test this?

You could try http://seyrsnys.myzen.co.uk/inav_dev/

mtnmikev, 2021-05-31:
Well done guys! Tested and working.

PR #7047 (2021-05-31): Set MAVLink v1 flag correctly
Fixes #7037 (maybe)

**TO BE CLEAR: I haven't had time to test if this works.**

@stronnag @mtnmikev @wx4cb Please could you all test if this fixes the MAVLink V1 issues?