iNavFlight/inav #7053: Suppor for Skystars F405HD mini

beeonlinecy, 2021-05-31:
## Current Behavior
No official support

## Desired Behavior
Support multirotors

## Suggested Solution
There is already a working target done and tested by me.

## Who does this impact? Who is this for?
Those who want to get a cheap good quality F4 flight controller.

## Additional context
This FC is a 20x20 small, cheap and good quality, having 
-6 uarts , 
-4 outputs, 
-current input, 
-i2c bus
-8Mb blackbox
-OSD
-Baro
It is confirmed on my test 5" multirotor(quad) with fully working gps,mag and baro.

balist911, 2021-06-03:
Nice idea. Will make many people happy.

beeonlinecy, 2021-06-03:
> Nice idea. Will make many people happy.

If there is anybody there who already have this FC and want to try my firmware build I am not greedy to share it

balist911, 2021-06-03:
I'll be very thankful. balist911@gmail.com

beeonlinecy, 2021-06-03:
forgot to tell you:
this is only tested with quads, if you are planning to try airplanes , be
careful it may burn your servo. Maybe i should build another version
separately for planes with different pinout one day

balist911, 2021-06-03:
I need firmware for a quadcopter. I won't use it on an airplane. Thanks

closed 2022-01-28 by DzikuVx