iNavFlight/inav #7060: Calibration Problems on INAV 3.0

CorruptName, 2021-06-01:
After doing the Calibration Process (both in the plane and on the table) it shows the model with the starboard wing slightly up when I hold the model level. Is this normal? Should I adjust accgain manually in cli until it is flat? What am I doing wrong?
I have this issue with 2 different wings, 1 with F405-Wing and the other is F765-Wing.
I even changed out the F765-Wing because I thought something was wrong with it after MANY calibration attempts.

Wing: AR Wing Pro
Version: INAV 3.0 RC2
Board: Matek F405-Wing & F765-Wing
Dump :[INAV_cli__WDE_20210601_150718.txt](https://github.com/iNavFlight/inav/files/6580053/INAV_cli__WDE_20210601_150718.txt)

breadoven, 2021-06-06:
Calibration doesn't level the board relative to the plane. You need to adjust that in the Configuration tab, Sensor alignment, Roll adjustment in your case.

avsaase, 2021-06-06:
Board alignment other than on yaw was removed from the configuration tab in 3.0. Use `align_board_roll` (in 0.1 degree units) in the CLI to set the board alignment.

CorruptName, 2021-06-06:
> Calibration doesn't level the board relative to the plane. You need to adjust that in the Configuration tab, Sensor alignment, Roll adjustment in your case.

> Board alignment other than on yaw was removed from the configuration tab in 3.0. Use `align_board_roll` (in 0.1 degree units) in the CLI to set the board alignment.

But if I'm calibrating with the FC inside the plane, why wouldn't it show the plane as level? In other words, why wouldn't it think it's level anyway? Is there something else it's using to calibrate besides the values I'm holding it at?

CorruptName, 2021-06-06:
If this is truly how it is intended to work it seems like there are probably a lot of aircraft flying around (at least) slightly crooked since it would be very hard for a person install the FC in the craft perfectly level and square.

CorruptName, 2021-06-06:
Please know that I don't mean to come across rude here, Just trying to understand. I'm aware that there is a ton of work involved here by many people and for that I am thankful. Maybe I should just clone and start digging into code to see what's going on and how I can improve it.

closed 2021-11-14 by DzikuVx