iNavFlight/inav #7062: FS-i6x does not working with FURYF4OSD

kotikvadik, 2021-06-02:
I have a Mamba F405 mk2 flight controller. Target firmware FURYF4OSD. I also use the Flysky FS-i6x remote control. FURYF4OSD firmware perfectly sees the remote in Betaflight. If I use INAV with MAMBA F405 board version 2.6.1 everything works fine too. But if in INAV I select the FURYF4OSD board with any firmware, for example, v 2.6.1, there is no reaction to the movements of the remote control sticks. What would you suggest.

pranjalk197, 2021-06-06:
Same issue from INAV 3.0 RC2 with FLYSKY FS IA6b (IBUS) on MATEK F722 STD.

SpencerMKSmith, 2021-06-09:
I'm having this issue also with my recently purchased Mamba F405 mk2 stack.

closed 2021-06-20 by DzikuVx

kotikvadik, 2021-06-20:
This does not work with FURYF4OSD! When I upload the firmware for the MAMBA F405, everything works fine.