iNavFlight/inav #7066: Blackbox recorded data is incorrect if no motors in mixer

gcmcnutt, 2021-06-03:
## Current Behavior
I was having trouble decoding blackbox data with my 2.6.1 configuration. After troubleshooting, I found that if no motors are in the mixer then the blackbox headers don't emit motors (count = 0). However, the actual blackbox data stream always emits motor0 data. This causes data to violate the data stream contract and the various decoders/viewers don't work.

## Steps to Reproduce

1. Create a configuration
2. Remove all motors (in my test case I had RCthrottle directly tied to a servo in the servo section, so didn't need motors)
2. Record data
3. Note that the downloaded blackbox data-stream includes an extra field for motor0
4.

## Expected behavior

For this case, the system probably should support no motor data (or some other error to note an invalid configuration).

## Suggested solution(s)

blackbox could conditionally emit motor0 based on count, like it does for motors>0

## Additional context

---

- FC Board name and vendor: Matek F722 Mini
- INAV version string: # INAV/MATEKF722MINI 2.6.1 Jun  1 2021 / 19:19:13 (64249b1a)

stronnag, 2021-06-03:
Please provide a log for 2.6.1 or 3.0 that exhibits the alleged problem. The log in #6214 decodes correctly (here is no motor in the `blackbox_decode` output.

gcmcnutt, 2021-06-04:
Here's the hex data -- see the variable byte array starting at 0xe6d

```
00000e00  5f 67 79 72 6f 5f 71 3a  35 30 30 0a 53 81 88 08  |_gyro_q:500.S...|
00000e10  98 84 8a 07 05 45 00 00  00 00 00 fe 07 ae 08 c3  |.....E..........|
00000e20  13 c3 13 c3 13 c3 13 c3  13 c3 13 c3 13 c3 13 f1  |................|
00000e30  dd c3 d0 0f 25 49 00 ca  aa e3 75 00 00 00 00 00  |....%I....u.....|
00000e40  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 b8  |................|
00000e50  17 b4 17 b8 17 b6 0f 00  00 00 00 fe 09 86 07 01  |................|
00000e60  00 00 e1 04 a4 02 da 1e  56 a0 01 a0 01 99 f7 ff  |........V.......|
00000e70  ff 0f 00 43 1b e7 07 a2  01 00 00 00 00 00 00 00  |...C............|
00000e80  00 00 00 00 00 02 d0 0f  c8 01 00 00 4b 00 00 19  |............K...|
00000e90  15 00 09 00 00 00 00 00  00 00 45 00 ca ce d6 75  |..........E....u|
```
That is motor0 output.  And the mixer had no motors at that time (didn't need them as I had raw rc throttle going to ESC). The headers declared 77 fields, but 78 are sent.

From blackbox.c:
```
    //Motors can be below minthrottle when disarmed, but that doesn't happen much
    blackboxWriteUnsignedVB(blackboxCurrent->motor[0] - getThrottleIdleValue());

    //Motors tend to be similar to each other so use the first motor's value as a predictor of the others
    const int motorCount = getMotorCount();
    for (int x = 1; x < motorCount; x++) {
        blackboxWriteSignedVB(blackboxCurrent->motor[x] - blackboxCurrent->motor[0]);
    }
```

[blackbox_log_2021-06-01_192817.TXT.zip](https://github.com/iNavFlight/inav/files/6595013/blackbox_log_2021-06-01_192817.TXT.zip)

closed 2024-02-02 by DzikuVx