iNavFlight/inav #7076: Auto Trim issues on a tail dragger

wx4cb, 2021-06-04:
i'm trying to understand how the continous auto trim is working with regards to a tail dragger.

I went out the morning with my grand tundra... flight went great, did a couple laps in manual, then switched to angle mode and let it sit there and fly around in angle mode so that it could trim the aircraft.

plane was a little nose heavy by about 3mm from the CG marks as this was  the maiden, but otherwise was balanced pretty well.

my issue is when i landed, (im pretty sure i was in manual mode but could be mistaken) the elevator was almost all the way pitch up (turns out to be about 1350uS), this of course is not acceptable as it is a tail dragger.

so my question is, any thoughts on what could have caused this?

I have included the 2 BB logs and also the pre tune diff and the post tune diff

For brevity, here's the DIFF of them

[image]

[INAV_cli_Tundra_20210529_160957.txt](https://github.com/iNavFlight/inav/files/6600976/INAV_cli_Tundra_20210529_160957.txt)

[after tune - INAV_cli_Tundra_20210604_153835.txt](https://github.com/iNavFlight/inav/files/6600979/after.tune.-.INAV_cli_Tundra_20210604_153835.txt)

[Grand Tundra BBLOG 6.4.2021.zip](https://github.com/iNavFlight/inav/files/6600987/Grand.Tundra.BBLOG.6.4.2021.zip)

wx4cb, 2021-06-04:
Here's the Resulting Outputs tab:

[image]

wx4cb, 2021-06-04:
Here are the raw DVR from the two flights today... now bear in mind, I *MAY* be wrong in how it's supposed to work and it may be working as designed, so this is more of a query than anything else

Tundra Flight 1: https://youtu.be/fpURk1xVhCg

Tundra Flight 2: https://youtu.be/aJhI378iVoQ

avsaase, 2021-06-05:
I added some extra restrictions for when the trims can be changed. I think they will solve this problem but I will have to test that this weekend.

avsaase, 2021-06-05:
To see how continuous autotrim adjusts the trims you can enable the autotrim debug mode (`set debug_mode = AUTOTRIM`) and put the debug osd element on your osd (`osd_layout 3 78 1 4 V` puts it on the third alternative OSD screen). You can then see the changing servo midpoints. A recording of that would be very helpful.

wx4cb, 2021-06-05:
@avsaase any chance you can do me a F405se target for the tundra and i can test on monday morning

charlie

avsaase, 2021-06-05:
[inav_3.0.0_MATEKF405SE.zip](https://github.com/iNavFlight/inav/files/6602473/inav_3.0.0_MATEKF405SE.zip)

Please flash with full chip erase. You can restore a `diff all`, except for those bad trim of course.

wx4cb, 2021-06-06:
@avsaase alex, can you do me one for the MATEKF722WPX target as well?

avsaase, 2021-06-07:
You should be able to find all target from this pull request here https://github.com/iNavFlight/inav/suites/2917574656/artifacts/65487247

wx4cb, 2021-06-07:
Here's the diff and BB log for the Mini Talon flight this morning. 

I know i did the auto tune wrong, forgot it was in manual mode, but my issues is after I come out of cruise mode, it did not want to climb at all and when i put it in manual on the turn just before landing it was almost all up elevator till landing.
if it's something i did wrong (outside of being in manual mode for the auto tune step lol my bad forgot that) let me know. happy to reset and go back when i put that new firmware on.

i've loaded the PR u p on that and the tundra and will get back out there asap. i unfortunately was running late this morning and needed to be at the airport for a meeting at 930.

DVR: https://youtu.be/xPcLMyMsSvM

[LOG00037.TXT](https://github.com/iNavFlight/inav/files/6611057/LOG00037.TXT)
[INAV_cli_MiniTalon_20210607_135048.txt](https://github.com/iNavFlight/inav/files/6611059/INAV_cli_MiniTalon_20210607_135048.txt)

avsaase, 2021-06-07:
It would be most helpful if you record the debug osd element while flying in acro/angle/cruise so we can see how the trims are changing. If the trim is way off it's probably best to reset them before the flight. 

The way I tested this feature is by briefly switching to manual with the sticks centered at random points throughout the flight. If the trim is off you'll immediately notice it pitching or rolling.

wx4cb, 2021-06-07:
@avsaase will do... next time i go out i will... not going to be tomorrow though..... have a 4am start

wx4cb, 2021-06-10:
@avsaase her you go alex.

here's the mini talon flight from this morning. forgot the wing spars of the tundra, but i'll try that one tomorrow.

https://youtu.be/Q-vibvjX23s

avsaase, 2021-06-10:
There are some things happening in the flight and the log that I cannot explain:

1. From the start of the flight the servo centers are 1435, 1435, 1423 and 1368 (?). Is that correct?
2. There are only 12 updates of the midpoints (top right value) during the entire flight. When all conditions are satisfied updates occur every 0.5 seconds so I expect this number to run into the hundreds assuming you do some hands-off flying.
3. The pitch I-term is maxed out at 165 for a significant portion of the flight. This implies the pitch trim is way off. Is this what you observed when you switched to manual at the end?

I will get some flights in with the updated code tonight.

wx4cb, 2021-06-10:
> There are some things happening in the flight and the log that I cannot explain:
> 
> 1. From the start of the flight the servo centers are 1435, 1435, 1423 and 1368 (?). Is that correct?

only thing i can thing of is that as i have trim always enabled, that when i threw it in auto launch it set those trims? I know that everything was set at 1500 before this flight.... I did throw it about 3 minutes earlier, but was a failed launch cus i messed up with my weak shoulder. (whole reason i have to use auto launch)

> 2. There are only 12 updates of the midpoints (top right value) during the entire flight. When all conditions are satisfied updates occur every 0.5 seconds so I expect this number to run into the hundreds assuming you do some hands-off flying.

i didnt do much "hands off" flying, I can do, but as I was by myself this morning i was trying to keep it within where i could catch it LOS if i messed up something

> 3. The pitch I-term is maxed out at 165 for a significant portion of the flight. This implies the pitch trim is way off. Is this what you observed when you switched to manual at the end?

not sure. wasnt paying attention for the most part was enjoying the flight. it did seem a little sluggish but i certainly wasn't yanking the elevator up like i did before. but the BBL should show where the sticks were.

i'm probably going to go out tomorrow. I want to try a different higher kv motor as i'm at 3/4 throttle for the most part. 

> I will get some flights in with the updated code tonight.

closed 2022-02-23 by DzikuVx