iNavFlight/inav #7077: Blackbox data corrupted

DarthPlasma, 2021-06-04:
I have problems in reading the blackbox data with inav 3.0. The data appear corrupted and the blackbox shows the model spinning randomly  at a rate of 50000-90000 deg/s

I tried different loggin rates (down to 1/32) with no success

wx4cb, 2021-06-06:
reformat the SD card and use the latest BB explorer

avsaase, 2021-06-06:
On my F405-WSE the logs are corrupted when I transfer them via msc but they work fine when I copy them via a USB SD-card reader.

wx4cb, 2021-06-06:
> On my F405-WSE the logs are corrupted when I transfer them via msc but they work fine when I copy them via a USB SD-card reader.

interesting... cus the logs for the tundra were transferred over msc (FC buried in the fuse) but i had to reformat it first using guiformatter and then it worked fine... I did notice though however, when the drive was loaded in windows the FAT table was what was corrupt.

PR #7086 (2021-06-08): Dzikuvx  potential fix for bb problems
hopefully 
closes #7084
closes #7077 

@stronnag could you please test? Looks good on my side
closed 2021-06-14 by DzikuVx