iNavFlight/inav #7078: Servo centering turns 90 degrees on boot

florind, 2021-06-04:
I am building a wing with Matek-411 WSE. iNav version 2.3.0
## Current Behavior
1. Plug in the battery
2. One servo unexpectedly turns 90 degrees while the other (correctly I think) only turns about 30 degrees. Both go to center position afterwards.
When I switch the servos around, the issue is present on the other servo.

## Expected behavior
When centering the servos, the movement should be less than 90 degrees.

- INAV version string: INAV/MATEKF411SE 2.3.0 Nov 24 2019 / 11:04:52 (d0bdd38c0)

rts18, 2021-06-06:
There was a feature implemented in 2.6.1 that fixes this problem. #6499

wx4cb, 2021-06-06:
i've had a similar thing.. was really weird.... the one eileron would go all the way down and the other would center, then it would just sit there sticking down. I even got as far as taking the servo out and testing it - not a mean feat on a grand tundra considering all the connectors.

weird thing was everything tested ok, and then i put everything back together and it still did it... , reflashed the board and it's fine...

strange one indeed

florind, 2021-06-06:
Interesting.I've upgraded to 2.6.0 with no change in behaviour. I reflashed then with 3.0.RC2 and this didn't make any difference either. Funny thing is that the initial bench test was just fine and this started happening after I've added the receiver, VTX and camera (I later removed all with no avail). Even stranger is that with the USB plugged-in, the servo centering works just fine on startup.
I ordered another identical FC, waiting for the delivery now.

avsaase, 2021-06-06:
#6499 only changes the first signal sent to the servos to the configured midpoint instead of a fixed 1500us. Clearly in some cases another value is sent to the servos on boot. I've had this happen as well but I cannot reliably reproduce the problem.

florind, 2021-06-21:
Update: I've switched with a new Matek-411 WSE and two new servos. The servos (https://www.aliexpress.com/item/32904792827.html?spm=a2g0s.9042311.0.0.6d724c4dvvcXrF) are of type GH-S37D (I think D stands for digital) although the article type is `GH-S37A`. The servo model is identical to the servos I initially used.
The behaviour is identical with the one I described in this ticket. 
More details:
I use the `Flying Wing` mixer.
The centering signal happens only on the S3 servo.
When I switched to an Airplane mixer (I think)  to see if there's any change, both servos (S3 and S4) performed the 90 deg. centering move on startup.

I ended-up  enlarging  the servo arm hole to accommodate the 90deg. roll. If anyone has a better idea I'm keen on trying out.

wx4cb, 2021-06-21:
i'm running the release version of 3.0 and mine does this, but returns to 1500 (or thereabouts) after it's booted

florind, 2021-06-21:
Same here, the servo returns to the center position in my case as well. It's the bootup centering that is incorrect for S3 using the Flying Wing mixer.

closed 2023-10-27 by DzikuVx