iNavFlight/inav #7084: Increased frequency of corrupted logs in 3.0.0

stronnag, 2021-06-06:
Over the last few months, I've noticed an increased number of blackbox logs with major metadata corruption, rendering the log contents un-decodable (or at least untrustworthy). This occurs even at low logging rates and quality SD cards (e.g. premium Samsung).

Example, two logs recorded recently, one is fine, one has almost no `H xxxx` data, making derived data (e.g. voltage) unusable.

## Good Log Metadata Section:

```
H Product:Blackbox flight data recorder by Nicholas Sherlock
H Data version:2
H I interval:256
H Field I name:loopIteration,time,axisRate[0],axisRate[1],axisRate[2],axisP[0],axisP[1],axisP[2],axisI[0],axisI[1],axisI[2],axisD[0],axisD[1],axisD[2],fwAltP,fwAltI,fwAltD,fwAltOut,fwPosP,fwPosI,fwPosD,fwPosOut,rcData[0],rcData[1],rcData[2],rcData[3],rcCommand[0],rcCommand[1],rcCommand[2],rcCommand[3],vbat,amperage,BaroAlt,rssi,gyroADC[0],gyroADC[1],gyroADC[2],accSmooth[0],accSmooth[1],accSmooth[2],attitude[0],attitude[1],attitude[2],motor[0],motor[1],servo[0],servo[1],servo[2],servo[3],servo[4],servo[5],servo[6],servo[7],servo[8],servo[9],servo[10],servo[11],servo[12],servo[13],servo[14],servo[15],navState,navFlags,navEPH,navEPV,navPos[0],navPos[1],navPos[2],navVel[0],navVel[1],navVel[2],navAcc[0],navAcc[1],navAcc[2],navTgtVel[0],navTgtVel[1],navTgtVel[2],navTgtPos[0],navTgtPos[1],navTgtPos[2],navSurf[0]
H Field I signed:0,0,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,0,1,1,0,1,1,1,1,1,1,1,1,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,0,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1
H Field I predictor:0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,4,9,0,0,0,0,0,0,0,0,0,0,0,0,4,5,8,8,8,8,8,8,8,8,8,8,8,8,8,8,8,8,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
H Field I encoding:1,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,3,0,0,1,0,0,0,0,0,0,0,0,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
H Field P predictor:6,2,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,1,1,1,1,1,1,1,3,3,3,3,3,3,3,3,3,1,1,1,1
H Field P encoding:9,0,0,0,0,0,0,0,7,7,7,0,0,0,0,0,0,0,0,0,0,0,8,8,8,8,8,8,8,8,6,6,6,6,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
H Field H name:GPS_home[0],GPS_home[1]
H Field H signed:1,1
H Field H predictor:0,0
H Field H encoding:0,0
H Field G name:time,GPS_fixType,GPS_numSat,GPS_coord[0],GPS_coord[1],GPS_altitude,GPS_speed,GPS_ground_course,GPS_hdop,GPS_eph,GPS_epv,GPS_velned[0],GPS_velned[1],GPS_velned[2]
H Field G signed:0,0,0,1,1,1,0,0,0,0,0,1,1,1
H Field G predictor:10,0,0,7,7,0,0,0,0,0,0,0,0,0
H Field G encoding:1,1,1,0,0,0,1,1,1,1,1,0,0,0
H Field S name:flightModeFlags,stateFlags,failsafePhase,rxSignalReceived,rxFlightChannelsValid,hwHealthStatus,powerSupplyImpedance,sagCompensatedVBat,wind[0],wind[1],wind[2],IMUTemperature,baroTemperature,sens0Temp,sens1Temp,sens2Temp,sens3Temp,sens4Temp,sens5Temp,sens6Temp,sens7Temp,escRPM,escTemperature
H Field S signed:0,0,0,0,0,0,0,0,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1
H Field S predictor:0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1
H Field S encoding:1,1,7,7,7,1,1,1,0,0,0,0,0,0,0,0,0,0,0,0,0,1,0
H Firmware type:Cleanflight
H Firmware revision:INAV 3.0.0 (b27b8758) MATEKF405SE
H Firmware date:Jun  2 2021 20:19:53
H Log start datetime:2021-06-05T17:45:30.355+00:00
H Craft name:DOLPHIN
H P interval:1/200
H minthrottle:1076
H maxthrottle:1950
H gyro_scale:0x3f800000
H motorOutput:1076,1950
H acc_1G:2048
H vbat_scale:111
H vbatcellvoltage:33,35,42
H vbatref:1646
H currentMeter:210,330
H looptime:1000
H rc_rate:100
H rc_expo:10
H rc_yaw_expo:30
H thr_mid:0
H thr_expo:0
H tpa_rate:10
H tpa_breakpoint:1500
H rates:36,16,3
H rollPID:12,12,5
H pitchPID:12,12,5
H yawPID:20,5,0
H altPID:20,5,5
H posPID:50,5,8
H posrPID:0,0,0
H levelPID:20,15,75
H magPID:60
H velPID:0,0,0
H yaw_lpf_hz:0
H dterm_lpf_hz:40
H dterm_lpf_type:1
H dterm_lpf2_hz:0
H dterm_lpf2_type:1
H deadband:4
H yaw_deadband:4
H gyro_lpf:0
H gyro_lpf_hz:25
H gyro_lpf_type:1
H dynamicGyroNotchRange:1
H dynamicGyroNotchQ:250
H dynamicGyroNotchMinHz:30
H gyro_notch_hz:0,0
H gyro_notch_cutoff:1,1
H acc_lpf_hz:15
H acc_hardware:7
H baro_hardware:4
H mag_hardware:0
H serialrx_provider:2
H motor_pwm_protocol:6
H motor_pwm_rate:8000
H debug_mode:0
H features:810028242
H waypoints:0,0
H acc_notch_hz:0
H acc_notch_cutoff:1
H pidSumLimit:500
H pidSumLimitYaw:350
H axisAccelerationLimitYaw:10000
H axisAccelerationLimitRollPitch:0
H rpm_gyro_filter_enabled:0
H rpm_gyro_harmonics:1
H rpm_gyro_min_hz:100
H rpm_gyro_q:500
``` 

##  Bad log metadata section

```
H Product:Blackbox flight data recorder by Nicholas Sherlock
H Data version:2
H I interval:256
H Field I name:loopIteration,time,axisRate[0],axisRate[1],axisRate[2],axisP[0],axisP[1],axisP[2],axisI[0],axisI[1],axisI[2],axisD[0],axisD[1],axisD[2],fwAltP,fwAltI,fwAltD,fwAltOut,fwPosP,fwPosI,fwPosD,fwPosOut,rcData[0],rcData[1],rcData[2],rcData[3],rcCommand[0],rcCommand[1],rcCommand[2],rcCommand[3],vbat,amperage,BaroAlt,rssi,gyroADC[0],gyroADC[1],gyroADC[2],accSmooth[0],accSmooth[1],accSmooth[2],attitude[0],attitude[1],attitude[2],motor[0],motor[1],servo[0],servo[1],servo[2],servo[3],servo[4],servo[5],servo[6],servo[7],servo[8],servo[9],servo[10],servo[11],servo[12],servo[13],servo[14],servo[15],navState,navFlags,navEPH,navEPV,navPos[0],navPos[1],navPos[2],navVel[0],navVel[1],navVel[2],navAcc[0],navAcc[1],navAcc[2],navTgtVel[0],navTgtVel[1],navTgtVel[2],navTgtPos[0],navTgtPos[1],navTgtPos[2],navSurf[0]
H Field I signed:0,0,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,0,1,1,0,1,1,1,1,1,1,1,1,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,0,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1
H Field I predictor:0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,4,9,0,0,0,0,0,0,0,0,0,0,0,0,4,5,8,8,8,8,8,8,8,8,8,8,8,8,8,8,8,8,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
H Field I encoding:1,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,3,0,0,1,0,0,0,0,0,0,0,0,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
H Field P predictor:6,2,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,1,1,1,1,1,1,1,3,3,3,3,3,3,3,3,3,1,1,1,1
H Field P encoding:9,0,0,0,0,0,0,0,7,7,7,0,0,0,0,0,0,0,0,0,0,0,8,8,8,8,8,8,8,8,6,6,6,6,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
H Field H name:GPS_home[0],GPS_home[1]
H Field H signed:1,1
H Field H predictor:0,0
H Field H encoding:0,0
H Field G name:time,GPS_fixType,GPS_numSat,GPS_coord[0],GPS_coord[1],GPS_altitude,GPS_speed,GPS_ground_course,GPS_hdop,GPS_eph,GPS_epv,GPS_velned[0],GPS_velned[1],GPS_velned[2]
H Field G signed:0,0,0,1,1,1,0,0,0,0,0,1,1,1
H Field G predictor:10,0,0,7,7,0,0,0,0,0,0,0,0,0
H Field G encoding:1,1,1,0,0,0,1,1,1,1,1,0,0,0
H Field S name:flightModeFlags,stateFlags,failsafePhase,rxSignalReceived,rxFlightChannelsValid,hwHealthStatus,powerSupplyImpedance,sagCompensatedVBat,wind[0],wind[1],wind[2],IMUTemperature,baroTemperature,sens0Temp,sens1Temp,sens2Temp,sens3Temp,sens4Temp,sens5Temp,sens6Temp,sens7Temp,escRPM,escTemperature
H Field S sH rpm_gyro_min_hz:100
H rpm_gyro_q:500
```

### Unfounded speculation

This is a result of timing / task priority changes in 3.0.0 (e.g #6828). 
It is also notable that idle CPU usage is somewhat greater in 3.0 (vice 2.6 and earlier).

stronnag, 2021-06-06:
yes, probably a duplicate of #7077, but this one has some evidence.

DzikuVx, 2021-06-06:
Interesting. SD process is called as often as before. I will dig into this
a little

stronnag, 2021-06-06:
Pawel, thanks, entirely speculative from looking at the merge list for 'suspects'.
In all instances I've seen, we lose 1536b (3x512 DOS FS blocks) in the metadata section, and probably have greater corruption reported later, for the examples above from @OlivierC-FR:

Good Log:

```
Log 1 of 1, start 04:37.640, end 25:58.620, duration 21:20.979

Statistics
Looptime           1018 avg            2.7 std dev (0.3%)
I frames    4911  117.2 bytes avg   575805 bytes total
P frames    4910   84.0 bytes avg   412360 bytes total
H frames      40   10.0 bytes avg      400 bytes total
G frames    6458   26.4 bytes avg   170375 bytes total
E frames       1    7.0 bytes avg        7 bytes total
S frames    9162   41.4 bytes avg   379246 bytes total
Frames      9821  100.6 bytes avg   988165 bytes total
Data rate    7Hz   1204 bytes/s      12100 baud

3 frames failed to decode, rendering 0 loop iterations unreadable. 
1247140 loop iterations weren't logged because of your blackbox_rate settings (1270970ms, 99.22%)
```

Bad log:

```
Log 1 of 1, start 02:27.663, end 19:44.977, duration 17:17.314

Statistics
Looptime           7859 avg        36705.6 std dev (467.0%)
I frames    3973  116.9 bytes avg   464287 bytes total
P frames     139   85.1 bytes avg    11835 bytes total
H frames      33   10.0 bytes avg      330 bytes total
G frames    5220   27.0 bytes avg   141169 bytes total
E frames       1    6.0 bytes avg        6 bytes total
S frames       9   43.4 bytes avg      391 bytes total
Frames      4112  115.8 bytes avg   476122 bytes total
Data rate    3Hz   1215 bytes/s      12200 baud

9372 frames failed to decode, rendering 4454 loop iterations unreadable. 1012721 iterations are missing in total (1033119ms, 99.60%)

WARNING: Missing expected metadata - check for log corruption
	Warning: No VBAT reference
	Error: No Firmware type metadata
	Error: No Firmware revision metadata
	Error: No P Interval
	Error: No Acc.1G metadata
```
(the latter some new warnings to go into blackbox_decode).

Logs generated within 15 minutes of each other. To quote Olivier *"Also, it's always been working for years, so it surprising it would fail now, that I uploaded to inav 3.0"*

DzikuVx, 2021-06-06:
The only change in the scheduler was that realtime priority has a real
priority. I wonder if that can cause a situation when 2 RT frames hog the
CPU for too long with no time for SD process. I will take another look at
the scheduler

DzikuVx, 2021-06-06:
The problem seems to be related not with the changed priority, but with te fact that there are less idle cycles now. This is interesting, will dig deeper into this

DzikuVx, 2021-06-06:
It looks like this:
1. BB is executed together with `PID` task and over here frequency have no really changed. We are running `PID` task at `2kHz` for years
2. `afatfs_poll();` was supposed to be executed during IDLE when no tasks are really selected by the sheduler
3. 3.0 has more tasks and it seems like there is a higher chance that IDLE phase is not happening because, well, there are more tasks executed all over again. 
4. The solution seems to be to lower the amount of tasks or allow the `afatfs_poll();` to be exucytet not only during IDLE phase

stronnag, 2021-06-06:
It's also the case that we writing much more data to the log than say 3 years ago, the header has grown by a factor of 2 at least.

DzikuVx, 2021-06-14:
I beliver we have this sorted for now

PR #7086 (2021-06-08): Dzikuvx  potential fix for bb problems
hopefully 
closes #7084
closes #7077 

@stronnag could you please test? Looks good on my side
closed 2021-06-14 by DzikuVx