iNavFlight/inav #7085: Build with VSCode

JulioCesarMatias, 2021-06-06:
after doing everything in the documents, the VSCode returns errors in the functions, but to compile and generate the .hex file it works normal. Is it normal to show errors like this?

[image]

JulioCesarMatias, 2021-06-06:
[image]

shellixyz, 2021-06-06:
It is probably not finding some headers. Might work better if you cmake configure the project from VSC

JulioCesarMatias, 2021-06-07:
I couldn't solve

JulioCesarMatias, 2021-06-14:
closing without solution

closed 2021-06-14 by JulioCesarMatias