iNavFlight/inav #7091: horizon swam away after dive

seag86, 2021-06-07:
Horizon completely swam away on wing after long dive in acro mode, but stabilized in flat flight. If there board auto trim in 3.0 firmware, can I disable it? 
https://drive.google.com/file/d/1ldMT7342HAIjSbFZZJsrMrFmYOwsX5Na/view?usp=sharing

JulioCesarMatias, 2021-06-07:
maybe it's the same reason for this other problem: https://github.com/iNavFlight/inav/discussions/7089

seag86, 2021-06-08:
Problem not just in osd, the wing doesn't fly in Angle mod

b14ckyy, 2021-06-08:
Please follow the guideline if you open a issue here. Your post gives us ZERO information. Post a video. Settings dump, maybe blackbox log if available, what hardware etc, etc.  With the given information we can't help.

seag86, 2021-06-08:
added video with description

b14ckyy, 2021-06-08:
> 5:00 spin in angle mode after cruise because of wrong board trim
That was a stall. your Current went down to idle, there was no thrust in angle mode so it stalled and dropped
> 5:30 dive in acro mode
same as above. You stayed at 0% throttle for more than 6 minutes straight and only in autonomous modes INAV throttled up by itself. No wonder why you did not get back any control. 
> 9:10 RTH doesn't work properly cause of wrong board trim
Actually your Horizon was totally off due to the complete random flying before. Also your tune is totally messed so RTH was not even able to fly a proper turn and the horizon was not able to find itself again because RTH was going crazy.
> 11:20 trying to fly in acro mode, and board trim slowly becomes normal
Yes because finally after 7 minutes of terror, you fly straight and level so the horizon can re-align on the accellerometer. 

Fix your tune and don't forget to apply throttle when you are flying in angle mode, creating stalles. Then it will work as intended :)

seag86, 2021-06-08:
In 2.6 version I didnt face that problems with my flying style, so can I disable board autotrim in 3.0?

avsaase, 2021-06-08:
If you don't have AUTOLEVEL mode enabled the board alignment is not changed. And even if it was it is likely not the cause of your problems.

b14ckyy, 2021-06-08:
this has nothing to do with autotrim. Nor with board level. 
this is pure horizon drift caused by cintinously random movements of the plane causing the accellerometer to stay disabled for long time.

the main issue was that you have cut the throttle and the plane was permanently stalling. has nothing to do with 3.0 and would have happened with 2.6 too. Even worse. 

the final release will have another addition to fight horizon drift better and allow Accellerometer realignment also during irregular roll inputs like yours. This will maybe resolve the issue you encountered here. 
But learn better throttle management to avoid stalls in the first place.

seag86, 2021-06-08:
Zero trottle is not problem, my wing perfectly flys with it when board properly trimed. Ok, I'll roll back to 2.6 for while.

seag86, 2021-06-09:
With 2.6 same problem, 2.5 better

avsaase, 2021-06-09:
What you are experiencing here is what has come to be known as horizon drift. There is already an open issue about this (https://github.com/iNavFlight/inav/issues/2459) so I'll close this one.

closed 2021-06-09 by avsaase