iNavFlight/inav #7092: Mavlink serial RC removed after reboot?

Matthias84, 2021-06-07:
## Current Behavior
Hi, I'm pretty curious on getting MavLink as RC running, to connect my drone via Wifi. It is powered by an Omnibus F4 nano v5 FC and has a RPI zero as companion SBC.
The telemetry is already working fine, but mavproxy complains about missing RC, so you need to setup a serial receiver with Mavlink protocol. Every time I enable this serial RX to a UART, it is removed after save & reboot.

## Steps to Reproduce

1. Connect a PI to a FC UART
2. Setup FC UART to Serial RX and Mavlink telemetry
3. Hit save
4. (After reboot, the UART is again unset)

## Expected behavior
Mavlink RC and Telemetry enabled

## Suggested solution(s)

## Additional context
https://pastebin.com/DwTEZv4a

---

- FC Board name and vendor: Omnibus F4 Nano v5
- INAV version string: INAV/OMNIBUSF4V3 3.0.0 May 28 2021 / 17:29:52

stronnag, 2021-06-07:
> 2. Setup FC UART to Serial RX and Mavlink telemetry

Configuration error, you cannot have Mavlink and RX on the same UART

closed 2021-06-07 by stronnag

Matthias84, 2021-06-07:
@stronnag  Thank you for clarification :smile: 

So if I want one Mavlink link that gives telemetry status and receives RC commands, how do I have to connect it? One UART for mavlink TX and one for mavlink RX?
_(sorry for asking, but I thought this is a pretty common usecase with just one bidirectional link)_

MorrisR2, 2022-01-18:
Inav now supports Mavlink RC and telemetry on the same port.
https://github.com/iNavFlight/inav/pull/6738

RomanLut, 2022-07-24:
Make sure you enabled "Telemetry output" on Config time - it took me a while to setup Mavlink RC!