iNavFlight/inav #7099: Possible problem with "slowDownForTurning"

breadoven, 2021-06-09:
Flew a short twisty 4 WP test mission the other day with a 5" multirotor which should have ended in hover position hold (basic final WP). It flew the mission as expected without slowing at the turns because `nav_mc_wp_slowdown `was set OFF. But rather than stopping and hovering at the last WP it started large increasing toilet bowling rotations so I switched off WP and changed to Position Hold. It stopped immediately and held position within a few seconds. I did notice the heading was changing by +- 10 degrees during the toilet bowling even though the fpv view "appeared" to be roughly pointing in the same direction. Checking afterwards heading does vary with pitch and roll (roughly 15 deg @ 45 deg P/R) even though the Mag alignment is set correctly which made me wonder if this was the problem. However, Poshold, RTH and WP with RTH landing had worked fine before this WP mission and selecting Poshold whilst it was toilet bowling stopped it immediately which makes me think the Mag issue isn't the problem.

I did however notice the following change related to `nav_mc_wp_slowdown`:
https://github.com/iNavFlight/inav/blob/9df51dbbbb99bc23bac4155dd18d002136eda098/src/main/navigation/navigation_multicopter.c#L448

Given that `nav_mc_wp_slowdown `was set OFF during this mission I'm wondering if this change might have caused the toilet bowling ? Haven't had a chance to test again with this set ON to see if it makes any difference.

Can upload log and video for further analysis if there's no obvious issue with the code.

breadoven, 2021-06-09:
Video https://www.youtube.com/watch?v=8uXpH5LeTJ0

Ignore the Compass Error, does nothing useful at the moment.

breadoven, 2021-06-13:
Made another test related to this problem and it seems to confirm this is something to do with nav_mc_wp_slowdown. Ran the same WP mission twice with the setting OFF and then ON, no other changes made. This was a WP mission with a stable approach to the last WP where it should have hovered.

With nav_mc_wp_slowdown = OFF it immediately started an increasing amplitude rolling, pitching drunken oscillation when the last WP was reached. With nav_mc_wp_slowdown = ON it behaved as expected, just settled down into a stable hover at the last WP. Something related to this setting does seem to be causing a problem at the final WP @DzikuVx .

Flight Logs attached, 10 = OFF and 11 = ON.

[blackbox_log_2021-06-13_200300.txt](https://github.com/iNavFlight/inav/files/6644472/blackbox_log_2021-06-13_200300.txt)

robertha64, 2021-06-14:
I had exactly same issue but with old 2.1 FW, I did find out it related to the speed at which final WP is aproached. When speed is 5m/s it will not happen, if speed is 13m/s the the WERY wild toilet bowling at last WP always happen. Looks like the proper Position hold is completely different from what happenning on last WP (which should be Unlimited Pos hold actually).

breadoven, 2021-06-15:
OK pretty sure now this is my mistake related to https://github.com/iNavFlight/inav/pull/6938. Resets `posControl.activeWaypointIndex` to 0 at the final WP finish which affects `isApproachingLastWaypoint` changing the way slowDownForTurning works during the final WP loiter. PR 6938 needs fixing.

closed 2021-06-15 by breadoven