iNavFlight/inav #7100: Negative altitude turns into decimal Km

Mateyhv, 2021-06-09:
Negative relative altitude counts in 1 meter increments down to -99m then switches to -0.1km, eg: 0.1km scale which renders the altimeter almost useless as the resolution drops down to 100 meters.

For negative relative altitude the switch should be at -999m to a 0.01km resolution scale, the same as for positive altitude.

Tested on 3.0 RC2 but possibly inherited from 2.6.1. I suspect the cause for the current behaviour is the limit in characters because of the added '-' sign.

Thanks

avsaase, 2021-06-09:
This boils down to a tradeoff between having the more granular altitude display for negative altitude <-100m and the need to permanently reserve an additional character position for this relatively rare scenario.

Mateyhv, 2021-06-09:
I agree the additional character is not a good idea. Can be done keeping the current characters count by representing -100m as -.10km which yelds a 10m resolution down to -999m which is much better than having a 100m resolution.

The scenario is on top of a hill/mountain flying down to the valley and back to the hill top. In the current situation when the altimeter shows -0.1km there is no way to know if the UAV is at -100m or at -199m which is very important to plan the climb back to the top.

avsaase, 2021-06-09:
So your proposed solution is to hide the leading zero before the decimal dot? This would require a change in `osdFormatCentiNumber()` and possibly also in the font files because only half the decimal is part of the digit following the dot.

Mateyhv, 2021-06-09:
Yes, the easy solution is to get rid of the zero in front of the dot. There are a few other possible solutions but all imply different grades of modifications and compromise which, given that going below -99m is not a common scenario, will not truly pay off.  

As you pointed out the construction of the dot is not particularly suitable here. Half a dot wont look very good but the benefit is still having accurate altitude information. The prettiest solution might be to add a minus character including the half dot although that means updating all fonts (and not sure how much of the code).

The altimeter is one of the six main flying instruments and as such should always give information to the pilot as accurate as possible.

breadoven, 2021-06-09:
How about changing the text attribute to TEXT_ATTRIBUTES_ADD_INVERTED (used for the "UNABLE TO ARM" message) for negative numbers. Display the "-" down to -99m then drop it thereafter. Might actually be useful displaying -ve altitudes with the different text attribute anyway.

Mateyhv, 2021-06-09:
Dropping the - will nicely max out the resolution. My main concern is that in some cases it can induce spatial disorientation and potentially create confusion, for example if a pilot is not aware of that feature and is flying at -40m after a while unexpectedly sees (positive) 105m. Without known visual references he has no way to tell if a strong thermal has risen the UAV or it is still keeping negative altitude. There will be also a weird relationship between altitude an variometer which will sum to the confusion. All in all I strongly suggest to keep the minus sign even if that means not having the highest ltitude resolution.

breadoven, 2021-06-10:
Well I set it up to use `TEXT_ATTRIBUTES_ADD_INVERTED` when negative and changed it so the altitude is set positive after the text attributes have been set (BLINKING is also applicable for alarms) so the formatting always shows the positive format, i.e. no negative sign at all. Negative is indicated only by `TEXT_ATTRIBUTES_ADD_INVERTED `. I would be happy to use this instead but I can understand others would prefer the minus sign in front to avoid confusion.

Probably easier to simply add the extra character rather than use truncated formats.

OptimusTi, 2021-06-10:
I'm pretty sure some boards can't work with inversed characters.

If a number is negative it needs to be displayed as such.

If a function is formatting the number to a useless number then the logic thing would be to  create a new function that shows the correct format.

Jetrell, 2021-06-10:
It certainly is confusing when going below -100m.  
I think the cleanest method, requiring no pilot deciphering. Would be to add a 4th character position for negative altitude's.  -`999m` 

 @avsaase I do see your point concerning keeping the character uniform. Depending in what direction the units would shift when another character place is added. Who knows what side of the screen an individual will choose to place the altitude readout.

avsaase, 2021-06-10:
The fix is trivial, replace `3` (the length of the number) with `alt < 0 ? 4 : 3` in 

https://github.com/iNavFlight/inav/blob/9df51dbbbb99bc23bac4155dd18d002136eda098/src/main/io/osd.c#L467

This requires a small configurator change to avoid placing the element in a position where it overlaps with something else when altitude < -99m.

Mateyhv, 2021-06-10:
Hands down the best solution! Thanks everyone

closed 2021-06-10 by Mateyhv