iNavFlight/inav #7105: Suggestion: Rename auto level

MrD-RC, 2021-06-11:
The auto level mode is great. However, the name seems to be confusing people if it’s function. I know in an ideal world, people will read the documentation to see what it does. But in reality, there are people assuming that this mode adds auto levelling to their selects flight mode. 

I suggest renaming the mode **auto board align** or **auto pitch align**, something like that.

Michael-Ringstrasse, 2021-06-12:
Great idea

avsaase, 2021-06-12:
I hope for 3.1 we can get rid of the mode altogether and add it as a feature that permanently enables it in angle and horizon mode.

MrD-RC, 2021-06-12:
@avsaase that would be the shizz right there!

Then we just need auto tune to automatically detect when the tune is crap and fix it 😉 Joking of course. But the feature idea for auto pitch alignment I like a lot.

avsaase, 2021-06-12:
I'm not a 100% sold on autolevel I must say but it's something to consider and collect more user experiences on.

avsaase, 2021-06-13:
I can think of two ways of thinking about the auto pitch trim functionality:

1. It is a feature that you only need during the initial setup of your plane, or when you made significant changes. In this case we could add it to the autotune procedure: the pitch trim is adjusted when autotune is enabled in angle/horizon mode and the sticks are centered. When you only want to change the pitch trim and not the gains/rates you just don't bang the sticks.
2. You want this permanently enabled so that angle/horizon mode is a pseudo altitude hold. Changing the mode to a feature would make sense in this case. If we go this route we should decrease the trim gain to avoid changing the pitch trim due to temporary disturbances such as thermals. 

I prefer the first option. 

Of course it's also possible to keep things as they are and just rename the mode and add some indication on the OSD that it's active (maybe display `ANGLE*` or `ANGLE+`?). This is more flexible but I don't like having so many mode modifiers that clutter the modes tab and require switches and channels.

b14ckyy, 2021-06-15:
it makes sense to permanently enable it as a feature. Ardupilot does basically the same.
but i am also fine with using it as a tuining mode only. But  in this case I would say to combine it with Autotrim. Not autotune.
autotune is exclusively PIFF tuning and people are advised to use it in ACRO only. As this will give best results.

If Continous Servo trim is enabled as a feature, then just the servo trim is disabled in the "AUTOTRIM" tuning mode but the board trim is done alone. 

in my opinion it makes sense to even force people into angle mode to do a manual Autotrim (even for servos) because if the plane slightly climbs or descents at a constant throttle value in acro mode, this changes airspeed and therefore the trims for the servos. And as the classic autotrim is a one time thing, it is important to have them on spot and consistent as they won't adjust later on. 

Continous trim will permanently adjust itself though. 

so to wrap this up: 

AUTOTRIM Mode = Servo trim + AoA Trim = forced Angle/Horicon mode

If Continous Servo Trim is enabled: 

AUTOTRIM Mode = AoA Trim only = forced Angle/Horicon mode

And Autotune stays Autotune as it is. Wasn't there even someone with the idea to pop up a system message if Auto tune is enabled in any other mode than ACRO?

avsaase, 2021-06-15:
I agree that making it part of autotrim makes more sense than autotune but I don't like forcing angle/horizon for autotrim. It works just fine in acro when flying level.

b14ckyy, 2021-06-15:
yeah maybe not force it but some clear hint that autolevel has to be done in angle mode though.

DzikuVx, 2022-05-30:
I think people got used to the name

closed 2022-05-30 by DzikuVx