iNavFlight/inav #7112: No way to setup OSD for DJI on SP RACING F4 EVO

pmarkiewicz, 2021-06-13:
## Current Behavior
I flashed v 3.0 to SP Racing F4 EVO to use one with DJI (there is no OSD chip on this board)
I cannot setup OSD on OSD page and get error while trying to setup OSD from cmd line.
OSD is enabled in configuration, DJI is setup in ports

## Expected behavior
Lack of OSD chip shouldn't prevent OSD setup for DJI

## Suggested solution(s)
For current situation OSD configuration should be available based on configuration tab, not on board features

INAV/SPRACINGF4EVO 3.0.0 Jun 12 2021 / 13:06:27 (3c7b1b7c)
[INAV_cli_AR_WING_20210613_205723.txt](https://github.com/iNavFlight/inav/files/6644636/INAV_cli_AR_WING_20210613_205723.txt)

Scavanger, 2021-06-14:
Your're right, since every FC can can used with DJI HD, OSD should be enabled for all targets.

Suggested solution:
Add `#define USE_OSD`  to target/common.h

wx4cb, 2021-06-14:
but the DJI system doesn't use the osd setup anyway, it gets all its information from MSP commands so there is nothing to setup.

You cannot control positions etc. anyway. only thing you can do is to use the CRAFT_NAME trick for things.

Scavanger, 2021-06-14:
@wx4cb That is completely wrong. Not all elements are supported, but they are set up in the same way as with the analogue OSD, via drag and drop in the Configurator or via the osd_layouts command.

wx4cb, 2021-06-14:
> @wx4cb That is completely wrong. Not all elements are supported, but they are set up in the same way as with the analogue OSD, via drag and drop in the Configurator or via the osd_layouts command.

then i stand corrected... i thought you couldn't change anything in there

PR #7118 (2021-06-20): Enable OSD for all targets
Enable OSD for all targets, so DJI HD OSD can be configured on targets without analogue OSD Chip.

fix #7112
closed 2021-06-20 by DzikuVx