iNavFlight/inav #7113: Add 'Adjustment' for 'Fixed Wing Level Trim' in INav 3.0

ShawnzAZ, 2021-06-13:
## Current Behavior
This does not look to be in-flight adjustable, as we used to do with board pitch angle in past versions.

## Desired Behavior
I would like to to see an OSD Element I can display on a tune focused OSD so that I can see the value adjusting as well as have it function like board angle pitch did in the past.

## Suggested Solution
## Desired Behavior
I would like to to see an OSD Element I can display on a tune focused OSD so that I can see the value adjusting as well as have it function liek board angle pitch did in the past.   I want to be able to tune where 'level' is vs having the trim fight the craft continuously.

## Who does this impact? Who is this for?
This is desired to make it so that I can field adjust and tune the craft to maintain altitude while in CRUISE / POSHOLD / RTH

## Additional context
This may be deemed unneccesary due to the new Continuous trim / auto Level functionality, but if so, why do we have the adjustment?

Many thanks INav devs!

wx4cb, 2021-06-14:
you can already have the angles on the OSD (unless i'm missing something). bottom right of attached pic.... pitch/roll and also board alignment pitch/roll 

being able to adjust it in flight would be cool though... as i could use the "spare" trim switches for it :D

[image]

ShawnzAZ, 2021-06-27:
They don't want us to adjust board angle in 3.0.   they want us to use 'fixed wing level trim'.

This is not adjustable in flight nor via the OSD menu.

ShawnzAZ, 2021-06-30:
After digging a little deeper, it seems that the in flight adjustment is to use the 'autolevel' mode and angle or horizon and let the FC figure it out vs setting a value.

I'd still like to be able to set it myself in flight (And see the value in the OSD) like I could before but I'm going to go out next time and get a good autotune session going, then autolevel  + horizon for a but to see if it can figure out the sweet spot for 'fixed wing level trim'.

PR #8028 (2022-05-14): Added FW level trim to in flight adjustments
Fixes #7729
Fixes #7113

Configurator PR: https://github.com/iNavFlight/inav-configurator/pull/1517

[Demo](https://youtu.be/K9SInOTxUTk)
closed 2022-05-14 by DzikuVx