iNavFlight/inav #7114: inav_3.0.0_MATEKF411SE_FD_SFTSRL1.hex impossible to set led strip

tonylm, 2021-06-14:
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

With this firmware (inav_3.0.0_MATEKF411SE_FD_SFTSRL1.hex), we don't have access to set led strip ... so I use inav_3.0.0_MATEKF411SE.hex and I do
## Steps to Reproduce

1.use inav_3.0.0_MATEKF411SE_FD_SFTSRL1.hex
2.on inav configurator , try to set led strip .. 
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor: mateksys F411 WSE
- INAV version string:

Scavanger, 2021-06-14:
Yes, LED pad is Softserial1 RX for this target. 
You have to decide: LED or full duplex softserial

tonylm, 2021-06-14:
ok, before (with Inav 2.61) it was on S6, so it was possible tu use a soft serial and led stip... 

Many thanks

closed 2021-06-14 by DzikuVx